UltimateHackingKeyboard/firmware #547: Initialization-related bugs

kareltucek, 2022-11-11:
- Macro output in macro events is lost because main update loop (including macro events) runs even before USB starts working.

- If first scancode originates from the macro engine, it is garbled. The reason is late reactive update of `usbBasicKeyboardProtocol`. (Report handling does not know yet that NKRO is used, and treats the memory as the boot protocol instead.)
  - macro reports need to get zeroed too when protocol changes
  - in order not to lose macro output, we need to update `usbBasicKeyboardProtocol` sooner (unlike standard keys, we won't get it entered again)

@benedekkupper / @mondalaci any ideas on how to detect when USB becomes ready?

benedekkupper, 2022-11-13:
Two things from my side:
We should probably store the macro reports in a fixed format, and upload the modifiers and scancodes into the usb report structure "manually" scancode-by-scancode.

I cannot think of a guaranteed way to detect that the host OS has fully initialized the interface, other than checking that the interrupt endpoint's data is being consumed. So we should test if we first send a completely empty report, and wait for that transfer to finish before sending any valid (macro) presses, whether this way can solve the problem.

mondalaci, 2022-11-13:
@kareltucek Let's use an NKRO buffer internally and dumb it down to a 6KRO buffer when 6KRO is used unless you have a better idea.

Let's test interface initialization as suggested.

kareltucek, 2022-11-13:
Thanks! Goid points!

benedekkupper, 2022-11-13:
There is a tradeoff to consider, storing macros in NKRO format allows more flexibility and complexity, but translating it to 6KRO is not always possible (rollover would happen after 6 key codes) - unless additional logic is added that splits one NKRO report into multiple 6KRO reports.

mondalaci, 2022-11-13:
@benedekkupper When 6KRO is used, the number of non-modifiers is intrinsically limited to 6 scancodes anyways, isn't it? Feel free to suggest a better way.

kareltucek, 2023-07-23:
I am out of ideas. 

- Tried to send an an empty (or even a multiple non-empty) reports and wait for basic interface to get a callback with `kUSB_DeviceHidEventSendResponse`. No results, those non-empty reports just don't produce any output. Not sure if there is a way to check that endpoint's data is actually being consumed rather than just successfully sent.
- I also tried to output the init sequence of USB callbacks and tried waiting until it is done: that is after all five report descriptors are queried. Also did not do the trick.

mondalaci, 2023-07-23:
How to reproduce this issue?

kareltucek, 2023-07-23:
Place a write text action into your $onInit and powercycle the keyboard.

kareltucek, 2023-07-23:
Or (with less clutter inbetween), place a code to send a scancode into your main.c somewhere between the initialization and start of the update loop. 

Here is my testing branch if it gives you a better idea: https://github.com/kareltucek/firmware/tree/init_issues

mondalaci, 2023-07-23:
Thanks! This issue is trivial to reproduce with the `write` command in `$onInit`.

@benedekkupper Any ideas which event / state variable we should watch for after which reports can be manipulated?

benedekkupper, 2023-07-24:
Ideally, starting to send reports should be good from the point when either a) the HID BOOT protocol is selected, or b) the HID report descriptor is read (only one of these is supposed to happen). This method is for all HID interfaces. If this doesn't work, then you can try to wait for the first OUT report (e.g. the keyboard LED setting message), which is limited to the keyboard interface, so the other interfaces might not be ready at that point (or not used at all by e.g. BIOS).

kareltucek, 2023-07-25:
Nope, neither seems to work.

I propose simply setting timer for 1000ms. (UHK won't produce anything significantly sooner after boot anyways.) If the initialization takes longer on some machines, it is not that big deal, since placing those write commands into `$onInit` is not a common usecase.

Regarding switching protocol during runtime, I propose to simply zero those macro reports. It may produce a behavior that is technically incorrect, but I can't imagine who would need to preserve scancode states across protocol switches. (If anyone minds, we can always implement a better solution.)

kareltucek, 2023-07-25:
Namely I propose this refactor: https://github.com/UltimateHackingKeyboard/firmware/pull/652

PR #652 (2023-07-29): Fix some initialization-related bugs.
Closes #547
closed 2023-07-29 by mondalaci