UltimateHackingKeyboard/firmware #549: Multi-device support

kareltucek, 2022-11-13:
@mondalaci do you have some specific plan for multi-device support in firmware?

I don't care much about Agent multi-device support, but I would appreciate being able to flash one UHK while having connected another UHK...

mondalaci, 2022-11-13:
Regarding firmware flashing, the connected devices should be listed by their name, type, and unique id, and the user should be allowed to pick the one to be flashed both in Agent and in the command line.

No firmware support is needed for this feature, so I'm closing this issue, but feel free to follow up.

kareltucek, 2022-11-13:
So the flashing script just needs to be altered to ask in tty which device to flash?

mondalaci, 2022-11-13:
Yes, but I'm unsure how much work it takes to retrieve the mentioned information. Also, I'd not use interactive input in the terminal and would rather stick to script parameters.

kareltucek, 2022-11-13:
Normally I would agree, but in this case - making it interact properly with Makefiles sounds like a great deal of pain for both implementation and for the end user, so I would go the interactive way unless we hit some unexpected obstacle.

(Arguments can always be added/used for sake of automation.)

mondalaci, 2022-11-13:
I'm not against interactivity, but the script should detect whether it's attached to tty and if not, it shouldn't try to be interactive.

kareltucek, 2022-11-14:
> Yes, but I'm unsure how much work it takes to retrieve the mentioned information.

Well, I see there no code for retrieving hardware config. The standard device-flashing scripts do not retrieve the user-defined name or unique id. So not straightforward with my knowledge of typescript and agent codebase.

However, it seems like persuading Agent that one of my UHKs does not exist should be quite straightforward...

mondalaci, 2022-11-14:
The `packages/usb` directory of Agent is a hodgepodge of random scripts. We'll eventually expose a nice TypeScript USB command library with command line utilities sitting on top of it. At that point, what you're after should be much easier to achieve and without later rewrites, so if this is not urgent, it's probably a better idea to wait.

kareltucek, 2022-11-14:
> so if this is not urgent, it's probably a better idea to wait.

Yes, indeed. In the meantime, deleting my uhk1 from model list of my firmware/lib/agent

kareltucek, 2022-11-14:
(Reopening was a misclick.)

closed 2022-11-14 by kareltucek