UltimateHackingKeyboard/firmware #550: Make tagging more robust

kareltucek, 2022-11-13:
I must have added the `v9.1.3` tag after creating the uploaded release, which unintentionally resulted in the current `55f4531` sha being used as the tag field in the package.json of the release. This makes Agent show the redundant and perhaps confusing sha for the 9.1.3 release. Still, Agent should be able to fetch the smart macro documentation when it's unavailable based on the sha, so it fulfills its purpose.

Do you see any other issues regarding the tag field?

For the sake of clarity, I think make-release.js should error out when encountering a tag that doesn't equal to `v${firmwareVersion}`, and allow for the `--allowSha` parameter to intentionally create a non-tagged release. Can you send us a PR for it?

Please create new issues in the future instead of commenting in existing ones. I think this is an unrelated issue.

_Originally posted by @mondalaci in https://github.com/UltimateHackingKeyboard/firmware/issues/545#issuecomment-1312698849_

kareltucek, 2022-11-13:
> Do you see any other issues regarding the tag field?

Well, in Agent, last module tag is not refreshed after flashing, but I believe that is a consequence of another known issue and is minor anyways.

I believe I had experienced some badly updated tags in Agent even for other modules, but I cannot reproduce it now, so nothing to worry about.

mondalaci, 2022-11-13:
We'll fix the Agent tag update issue.

PR #551 (2022-11-13): Require tag to conform to version, or an explicit override.
Closes #550.
closed 2022-11-13 by mondalaci