UltimateHackingKeyboard/firmware #554: Implementing gamepad controls

kareltucek, 2022-11-13:
@benedekkupper am I right in assuming that with 35048cce003f7b0f0bbf61a18ef244f71d230091 gamepad should be usable (when I manually throw in a `UsbGamepadSetProperty(ActiveUsbGamepadReport, GAMEPAD_A, 1);` call?

(So far tried with https://github.com/kareltucek/firmware/tree/gamepad_zerotest in SuperTuxKart (which detects the gamepad interface, but does not react to the A action, although it should according to its settings), and a bunch of online gamepad testers, with no luck.)

benedekkupper, 2022-11-13:
Are you testing with Windows or other OS?

kareltucek, 2022-11-13:
Linux by default.

Will try with Windows and suggested padding.

benedekkupper, 2022-11-13:
On Linux I used the `jstest` tool to check the functionality.

kareltucek, 2022-11-30:
Sorry about the delay.

Still no luck. Tried with linux+jstest, and with windows + some random native tester, both with the padding set to 6 and 0.

I have also force pushed the mentioned branch. I expect the branch to produce some "random" results when letter of the left UHK half are tapped - e.g., I believe that q and w should produce the gamepad A and B (although I didn't check that very thoroughly). If `~` is tapped, display should show `buttons` value of the report with some delay.

kareltucek, 2022-11-30:
I do see some packets on the gamepad endpoint when UHK is connected.

I don't see any packets when trying to send gamepad actions. In such case, `kStatus_USB_InvalidHandle` is returned from `UsbGamepadAction()`.

(I see mouse / keyboard events fine in wireshark.)
(Not sure whether still relevant, in any case, I am not sure how to swap endpoint numbers (while not breaking them at the same time).)

kareltucek, 2022-11-30:
I have filled in the missing 

```
    UsbCompositeDevice.gamepadHandle = UsbDeviceCompositeConfigList.config[USB_GAMEPAD_INTERFACE_INDEX].classHandle;
```

in right/src/usb_composite_device.c

Now I see packets coming in via wireshark.

Jstest still does not react. I will try to debug OS side...

kareltucek, 2022-11-30:
Ah, voila, working now. (Forgotten padding at 6.)

kareltucek, 2022-11-30:
Thanks for your help!

closed 2022-11-30 by kareltucek