UltimateHackingKeyboard/firmware #555: Secondary roles not usable on letter keys when typing fast

rwild003, 2022-11-15:
I'm trying to use the secondary role to have modifiers (ctrl,alt,etc) accessible from the home row. That means the standard letter keys are used for a secondary role.

It seems there is some kind of timeout active after shortly pressing a button with a secondary role: When typing normally the secondary role gets always triggered on all keys that have a secondary role mapped.

I type around 90-100WPM. If I slow down just a little bit everything works as it should. I guess with 50WPM there wouldn't be a problem. But there are people that type much faster than I...

WindChimeRan, 2022-11-15:
[A guide to home row mods](https://precondition.github.io/home-row-mods)

I found this blog relevant. I guess you'll need to customize the timeout through the advanced macros `resolveSecondary <timeout in ms>`.

Zot1881, 2022-11-15:
If i understand it correctly it's not just `resolveSecondary <timeout in ms>.`
It's more like this
```
resolveSecondary 150 10 base_action secondary_action
base_action:
pressKey space
break
secondary_action:
holdLayer mouse
break
```
so you need to write macro to use this functionality

kareltucek, 2022-11-15:
Also, to be precise:

> It seems there is some kind of timeout active after shortly pressing a button with a secondary role: When typing normally the secondary role gets always triggered on all keys that have a secondary role mapped.

There is no timeout. The secondary role kicks in whenever two key activations overlap.

ahidalgob, 2023-02-10:
@Zot1881 solution's good enough for me, and very simple to do :) thank you very much! For me `200 50` works good, fast enough to get almost no mistakes but slow enough to handle not-so-confident presses (using it as esc key / mod layer in the caps lock key). Thank you!

rafaelpirolla, 2023-10-06:
This was deprecated... Need to use ifPrimary advancedStrategy goTo ...
Unsure how to implement home row mods anymore...

kareltucek, 2023-10-06:
Only `resolveSecondary` was deprecated.

For instance:
```
ifSecondary advancedStrategy final holdLayer fn
holdKey d
```

Or you can `set secondaryRole.defaultStrategy advanced` in a macro named `$onInit` and then map the keys via key popover:

[image]
[image]

rafaelpirolla, 2023-10-06:
Yes... In the recent past I had:
```
postponeKeys delayUntilReleaseMax 200
postponeKeys ifPlaytime 200  final tapKey <<SHORTCUT>>
postponeKeys holdKey <<SHORTCUT>>
```
Now I have this $onInit:
```
set secondaryRole.defaultStrategy advanced
set secondaryRole.advanced.timeout 200
set secondaryRole.advanced.timeoutAction secondary
set secondaryRole.advanced.safetyMargin 0
set secondaryRole.advanced.triggerByRelease 0
set secondaryRole.advanced.doubletapToPrimary 1
set secondaryRole.advanced.doubletapTime 200
```
And I'm using the GUI to set the Secondary role... I need to read more about these configs to get it better.
Thanks for all the help @kareltucek, you're really helpful.

rafaelpirolla, 2023-10-06:
Can I override the timeout (and all these configs) for a specific key?
My Return with Shift is behaving strangely...

kareltucek, 2023-10-06:
- What is your return mapping? Just plain shift+enter?
- What is the problem with it?
- Whih exact configs would you like to override and why do you think it should help?

As to docs, are there any specific topics which should be covered and are not, or anything else that can be improved? (It is not that I do not care, it is that I am missing an insight into why the docs suck so much.)

rafaelpirolla, 2023-10-09:
1. Actually no, I said it wrong, Scancode Enter, Secondary role Mod layer.
2. Sometimes I want the Mod and it's interpreted as enter, I think I release too fast... I'll try with your example today but I believe the 200 will be the default:
```
ifSecondary advancedStrategy final holdLayer mod
holdKey enter
```
3. I have two guesses:
3.1. The 200ms, maybe I need the Layer to be activated faster on this key
3.2 It could also be that, since I'm using the Mod layer to move the cursor, that I'm pressing the h key (left arrow in mod layer) at the same time as the enter/mod key.

The topics are all there, and **the documentation is great**. I'm just not able to grasp the whole of it, somehow.