UltimateHackingKeyboard/firmware #560: make-release.js problems?

kareltucek, 2022-11-20:
When trying to use `make-release.js`, with node 16.16, I am getting:

```
> uhk-agent@2.0.2 convert-user-config-to-bin
> node -r ts-node/register ./packages/usb/user-config-json-to-bin.ts "/opt/firmware/scripts/uhk-firmware-9.1.4/devices/uhk60-right/config.bin"

TypeError [ERR_UNKNOWN_FILE_EXTENSION]: Unknown file extension ".ts" for /opt/firmware/lib/agent/packages/usb/user-config-json-to-bin.ts
    at new NodeError (node:internal/errors:372:5)
    at Object.getFileProtocolModuleFormat [as file:] (node:internal/modules/esm/get_format:76:11)
    at defaultGetFormat (node:internal/modules/esm/get_format:118:38)
    at defaultLoad (node:internal/modules/esm/load:21:20)
    at ESMLoader.load (node:internal/modules/esm/loader:407:26)
    at ESMLoader.moduleProvider (node:internal/modules/esm/loader:326:22)
    at new ModuleJob (node:internal/modules/esm/module_job:66:26)
    at ESMLoader.#createModuleJob (node:internal/modules/esm/loader:345:17)
    at ESMLoader.getModuleJob (node:internal/modules/esm/loader:304:34)
    at async Promise.all (index 0) {
  code: 'ERR_UNKNOWN_FILE_EXTENSION'
}
/opt/firmware/scripts/node_modules/shelljs/src/common.js:401
      if (config.fatal) throw e;
                        ^

Error: exec: TypeError [ERR_UNKNOWN_FILE_EXTENSION]: Unknown file extension ".ts" for /opt/firmware/lib/agent/packages/usb/user-config-json-to-bin.ts
    at new NodeError (node:internal/errors:372:5)
    at Object.getFileProtocolModuleFormat [as file:] (node:internal/modules/esm/get_format:76:11)
    at defaultGetFormat (node:internal/modules/esm/get_format:118:38)
    at defaultLoad (node:internal/modules/esm/load:21:20)
    at ESMLoader.load (node:internal/modules/esm/loader:407:26)
    at ESMLoader.moduleProvider (node:internal/modules/esm/loader:326:22)
    at new ModuleJob (node:internal/modules/esm/module_job:66:26)
    at ESMLoader.#createModuleJob (node:internal/modules/esm/loader:345:17)
    at ESMLoader.getModuleJob (node:internal/modules/esm/loader:304:34)
    at async Promise.all (index 0) {
  code: 'ERR_UNKNOWN_FILE_EXTENSION'
}

    at Object.error (/opt/firmware/scripts/node_modules/shelljs/src/common.js:110:27)
    at execSync (/opt/firmware/scripts/node_modules/shelljs/src/exec.js:102:12)
    at _exec (/opt/firmware/scripts/node_modules/shelljs/src/exec.js:205:12)
    at /opt/firmware/scripts/node_modules/shelljs/src/common.js:335:23
    at Object.<anonymous> (/opt/firmware/scripts/make-release.js:55:5)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:77:12)
```

When trying to use `make-release.js, with node 12.22.12 (as firmware/README.md instructs), but Agent built with 16.16, I am getting:

```
> uhk-agent@2.0.2 postinstall /opt/firmware/lib/agent
> lerna bootstrap

/opt/firmware/lib/agent/node_modules/@lerna/package-graph/index.js:79
            if (depNode?.version) {
                        ^

SyntaxError: Unexpected token '.'
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/opt/firmware/lib/agent/node_modules/@lerna/query-graph/query-graph.js:3:26)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! uhk-agent@2.0.2 postinstall: `lerna bootstrap`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the uhk-agent@2.0.2 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/me/.npm/_logs/2022-11-20T15_04_38_224Z-debug.log
```

Any ideas what is wrong? :-/

ert78gb, 2022-11-20:
Use node 16. I updated the readme.md of the firmware

kareltucek, 2022-11-20:
I *am* using node 16 (v16.18.1) - this version results in the `Error: exec: TypeError [ERR_UNKNOWN_FILE_EXTENSION]: Unknown file extension ".ts"` error above.

kareltucek, 2022-11-20:
Both agent and firmware/scripts reinstalled / rebuilt with node 16.16.0, still the same problem:

```
> uhk-agent@0.0.0 copy-to-tmp-folder
> node ../../scripts/copy-to-tmp-folder.js

lerna success run Ran npm script 'build' in 6 packages in 85.4s:
lerna success - kboot
lerna success - uhk-agent
lerna success - uhk-common
lerna success - uhk-smart-macro
lerna success - uhk-usb
lerna success - uhk-web
mkdir -p /opt/firmware/scripts/uhk-firmware-9.1.4/devices/uhk60-right
chmod 644 /opt/firmware/scripts/../right/uhk60v1/build_make/uhk60v1.hex
cp /opt/firmware/scripts/../right/uhk60v1/build_make/uhk60v1.hex /opt/firmware/scripts/uhk-firmware-9.1.4/devices/uhk60-right/firmware.hex
cp /opt/firmware/scripts/../right/uhk60v1/build_make/uhk60v1.map /opt/firmware/scripts/uhk-firmware-9.1.4/devices/uhk60-right/firmware.map
exec npm run convert-user-config-to-bin -- /opt/firmware/scripts/uhk-firmware-9.1.4/devices/uhk60-right/config.bin { cwd: '/opt/firmware/scripts/../lib/agent' }

> uhk-agent@2.0.2 convert-user-config-to-bin
> node -r ts-node/register ./packages/usb/user-config-json-to-bin.ts "/opt/firmware/scripts/uhk-firmware-9.1.4/devices/uhk60-right/config.bin"

TypeError [ERR_UNKNOWN_FILE_EXTENSION]: Unknown file extension ".ts" for /opt/firmware/lib/agent/packages/usb/user-config-json-to-bin.ts
    at new NodeError (node:internal/errors:372:5)
    at Object.getFileProtocolModuleFormat [as file:] (node:internal/modules/esm/get_format:76:11)
    at defaultGetFormat (node:internal/modules/esm/get_format:118:38)
    at defaultLoad (node:internal/modules/esm/load:21:20)
    at ESMLoader.load (node:internal/modules/esm/loader:407:26)
    at ESMLoader.moduleProvider (node:internal/modules/esm/loader:326:22)
    at new ModuleJob (node:internal/modules/esm/module_job:66:26)
    at ESMLoader.#createModuleJob (node:internal/modules/esm/loader:345:17)
    at ESMLoader.getModuleJob (node:internal/modules/esm/loader:304:34)
    at async Promise.all (index 0) {
  code: 'ERR_UNKNOWN_FILE_EXTENSION'
}
/opt/firmware/scripts/node_modules/shelljs/src/common.js:401
      if (config.fatal) throw e;
                        ^

Error: exec: TypeError [ERR_UNKNOWN_FILE_EXTENSION]: Unknown file extension ".ts" for /opt/firmware/lib/agent/packages/usb/user-config-json-to-bin.ts
    at new NodeError (node:internal/errors:372:5)
    at Object.getFileProtocolModuleFormat [as file:] (node:internal/modules/esm/get_format:76:11)
    at defaultGetFormat (node:internal/modules/esm/get_format:118:38)
    at defaultLoad (node:internal/modules/esm/load:21:20)
    at ESMLoader.load (node:internal/modules/esm/loader:407:26)
    at ESMLoader.moduleProvider (node:internal/modules/esm/loader:326:22)
    at new ModuleJob (node:internal/modules/esm/module_job:66:26)
    at ESMLoader.#createModuleJob (node:internal/modules/esm/loader:345:17)
    at ESMLoader.getModuleJob (node:internal/modules/esm/loader:304:34)
    at async Promise.all (index 0) {
  code: 'ERR_UNKNOWN_FILE_EXTENSION'
}

    at Object.error (/opt/firmware/scripts/node_modules/shelljs/src/common.js:110:27)
    at execSync (/opt/firmware/scripts/node_modules/shelljs/src/exec.js:102:12)
    at _exec (/opt/firmware/scripts/node_modules/shelljs/src/exec.js:205:12)
    at /opt/firmware/scripts/node_modules/shelljs/src/common.js:335:23
    at Object.<anonymous> (/opt/firmware/scripts/make-release.js:55:5)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:77:12)
```

kareltucek, 2022-11-20:
Also, there is a warning

```
> uhk-agent@0.0.0 build
> webpack && npm run install:build-deps && npm run download-firmware && npm run copy-to-tmp-folder

assets by path *.js 2.64 MiB
  asset electron-main.js 2.64 MiB [compared for emit] [minimized] (name: main) 2 related assets
  asset preload.js 178 bytes [compared for emit] [from: src/preload.js] [copied] [minimized]
assets by path *.json 529 bytes
  asset package.json 426 bytes [compared for emit] [from: src/package.json] [copied]
  asset manifest.json 103 bytes [compared for emit] [from: src/manifest.json] [copied]
orphan modules 503 KiB [orphan] 243 modules
runtime modules 786 bytes 4 modules
cacheable modules 5.4 MiB
  modules by path ../ 3.24 MiB 780 modules
  modules by path ./ 2.17 MiB
    javascript modules 2.13 MiB 478 modules
    json modules 37.3 KiB
      modules by path ./node_modules/ajv/dist/ 3.07 KiB 2 modules
      + 4 modules
+ 33 modules> uhk-agent@0.0.0 build
> webpack && npm run install:build-deps && npm run download-firmware && npm run copy-to-tmp-folder

assets by path *.js 2.64 MiB
  asset electron-main.js 2.64 MiB [compared for emit] [minimized] (name: main) 2 related assets
  asset preload.js 178 bytes [compared for emit] [from: src/preload.js] [copied] [minimized]
assets by path *.json 529 bytes
  asset package.json 426 bytes [compared for emit] [from: src/package.json] [copied]
  asset manifest.json 103 bytes [compared for emit] [from: src/manifest.json] [copied]
orphan modules 503 KiB [orphan] 243 modules
runtime modules 786 bytes 4 modules
cacheable modules 5.4 MiB
  modules by path ../ 3.24 MiB 780 modules
  modules by path ./ 2.17 MiB
    javascript modules 2.13 MiB 478 modules
    json modules 37.3 KiB
      modules by path ./node_modules/ajv/dist/ 3.07 KiB 2 modules
      + 4 modules
+ 33 modules

WARNING in ../uhk-smart-macro/node_modules/keyv/src/index.js 22:14-40
Critical dependency: the request of a dependency is an expression
 @ ../uhk-smart-macro/node_modules/cacheable-request/dist/index.js 9:0-24 183:42-46 208:36-40 212:29-33 218:29-33
 @ ../uhk-smart-macro/node_modules/got/dist/source/core/index.js 7:0-89 823:41-57 971:33-52
 @ ../uhk-smart-macro/node_modules/got/dist/source/index.js 14:0-32 14:0-32
 @ ../uhk-smart-macro/dist/download-tree-to-folder.js 5:0-22 26:23-33
 @ ../uhk-smart-macro/dist/download-smart-macro-doc.js 2:0-68 23:10-30
 @ ../uhk-smart-macro/dist/index.js 1:0-46 1:0-46
 @ ./src/services/smart-macro-doc.service.ts 8:0-56 64:18-39
 @ ./src/electron-main.ts 22:0-74 58:31-51

WARNING in ../uhk-smart-macro/node_modules/keyv/src/index.js 22:14-40
Critical dependency: the request of a dependency is an expression
 @ ../uhk-smart-macro/node_modules/cacheable-request/dist/index.js 9:0-24 183:42-46 208:36-40 212:29-33 218:29-33
 @ ../uhk-smart-macro/node_modules/got/dist/source/core/index.js 7:0-89 823:41-57 971:33-52
 @ ../uhk-smart-macro/node_modules/got/dist/source/index.js 14:0-32 14:0-32
 @ ../uhk-smart-macro/dist/download-tree-to-folder.js 5:0-22 26:23-33
 @ ../uhk-smart-macro/dist/download-smart-macro-doc.js 2:0-68 23:10-30
 @ ../uhk-smart-macro/dist/index.js 1:0-46 1:0-46
 @ ./src/services/smart-macro-doc.service.ts 8:0-56 64:18-39
 @ ./src/electron-main.ts 22:0-74 58:31-51
```

ert78gb, 2022-11-20:
What is the git sha of the agent submodule? If I reset the firmware repo and run the `git submodule update --init` that is in the `lib/readme.md` then it will clone to the `0a17555a929df6ca21fa637c02b9e0645bead14e` version of the agent. 2022-03-10 is the commit date of this sha.

why do you use this old version of the agent? I think worth to upgrade to the latest version of agent

kareltucek, 2022-11-20:
My lib/agent is checked out at `be722d65e784d9a2c1115c0cf4c8b8881b0b1119`.

kareltucek, 2022-11-20:
> why do you use this old version of the agent? I think worth to upgrade to the latest version of agent

🤔 I guess that everyone manages the lib/agent manually and no one bothers to update the git submodule references.

kareltucek, 2022-11-20:
Seems to work fine now!

ert78gb, 2022-11-20:
thx for the confirmation. I am just writing automated test to prevent this kind of error in the future

PR #561 (2022-11-20): doc: update the Node.js section of development setup
closes #560
closed 2022-11-20 by ert78gb