UltimateHackingKeyboard/firmware #563: autoRepeat with tapKeySeq doesn't repeat

rolandio, 2022-11-25:
Tried out some commands today and the following doesn't seem to work:

```
autoRepeat tapKeySeq CS-u 0 0 5 e space
```

I'm using a de_DE-keyboard layout, but also tried variants of the above in simpler forms, e.g.:
```
autoRepeat tapKeySeq a
```

The `autoRepeat` itself seems to be completely ignored in combination with `tabKeySeq`. So I wonder: is there a `tapKeySeq`-stop-sequence I am missing? Putting `final` in front of `tapKeySeq` did not work either. Or is it a bug?

The following command autorepeats fine:
```
autoRepeat tapKey a
```

The following is a workaround, that works for the moment, but is basically programming out an `autoRepeat`:
```
tapKeySeq CS-u 0 0 5 e space
ifNotReleased delayUntil 500
loop: ifNotReleased tapKeySeq CS-u 0 0 5 e space
ifNotReleased delayUntil 50
ifNotReleased goTo loop
```

kareltucek, 2022-11-25:
Probably a bug.

kareltucek, 2022-11-25:
But `autoRepeat tapKeySeq CS-u 0 0 5 e space` it works fine for me. (Produces `^^^^^^^^^^^^...` when held.)

Can you share your UserConfig? (Ideally just renamed as a txt.)

closed 2022-11-25 by rolandio

kareltucek, 2022-11-25:
Ok! Thanks for info!