UltimateHackingKeyboard/firmware #566: Cannot set navigation mode for ctrl layer: Unrecognized layer.: ctrl

googlegreg, 2022-11-29:
It looks like there's an issue with setting navigation modes with the ctrl layer.

I tried to have ctrl+touchpad zoom in & out (this is analagous to ctrl + mousewheel for me).

Typing it myself or using the editor I tried:

`set module.keycluster.navigationMode.ctrl zoom`

The LED display shows ERR and the resultant status shown from printStatus is:

`$onInit: 0: 2: Unrecognized layer.: ctrl zoom`

Thing I tried around this issue:

- Create `ctrl` layer (yes, at first I didn't have this layer so I thought this would fix it. It did not)
- Other variants of the command - they seem to work e.g. `set module.keycluster.navigationMode.base zoom` works just fine
- Turning it off and on again (obviously)

I'd be happy to be wrong but this smells like the `.ctrl` part is not being correctly parsed as a layer, especially since in-place editing yields exactly this command.

kareltucek, 2022-11-29:
Correct layer id is `control`.

(Yes, ctrl layer has to be defined if you want to map a navigation mode to it.)

But yes, `ctrl` is definitely more intuitive, so we will make the parser accept ctrl as well.

googlegreg, 2022-11-30:
Thanks for the reply. Maybe this is a documentation bug then. Also the
in-place editor creates the 'ctrl' string so I think that probably warrants
a fix.

googlegreg, 2022-11-30:
> Yes, on one abstraction level, it is a documentation bug. On another, it
> is a specification bug, because the id probably should have been ctrl from
> the start.
>
> Regarding the in-place editor, we have not implemented any dictionary for
> the monaco editor (yet), so the reason it suggest ctrl is that *you* have
> used it in your macro.

Thanks for clarifying!

closed 2022-12-20 by mondalaci