UltimateHackingKeyboard/firmware #569: backlight macros don't work

IzK666, 2022-12-03:
Using "set backlight.strategy constantRgb; set backlight.constantRgb.rgb 255 0 0" shows ERR on the lcd screen.
I've an UHK v1 with a cluster module and my intention was to change the cluster lightning depending on the keymap used.

kareltucek, 2022-12-03:
If you mean literally with a `;`, then that's the problem. 

```
set backlight.strategy constantRgb
set backlight.constantRgb.rgb 255 0 0
```

kareltucek, 2022-12-03:
But RGB light is not supported with UHKv1 as far as I know.

IzK666, 2022-12-03:
I tested both ways: with semicolon and without. I don't remember if one of them threw the err code or both.

That's what I suspected, that the cause might be the lack of lights of the v1.
Can you enable them? The cluster module leds can be controlled with the general "key backlight brightness" slider, so it might not be very difficult to enable it and imho can be useful.

kareltucek, 2022-12-03:
@mondalaci ?

mondalaci, 2022-12-07:
For me, ERR didn't appear; another command has likely caused it on @IzK666's side. But I can confirm that these commands have no effect on the UHK 60 v1.

Even though the UHK 60 v1 doesn't support RGB backlighting, the key cluster module does, so the above commands should work regardless. 

@kareltucek Would you please fix this issue?

kareltucek, 2022-12-07:
@mondalaci, what does that mean implementation-wise?

kareltucek, 2022-12-07:
I mean:

1) what is the actual specification for backlight of the right and left half of v1?

2) I suspect there may be some hardware differences between v1 and v2 right and *maybe* left half (?). I am not sure what these differences mean for implementation.

(Answer to 1 is required, any pointers regarding 2 are welcome, but I guess I should be able to figure them myself.)

mondalaci, 2022-12-07:
1. I'm unsure whether monochrome functional backlighting makes sense for the v1. Not lighting none actions up would probably be useful, but for the time being, I'd just light up every LED.

2. Different LED driver ICs are used for the v1 vs the v2, but we should only care about the relevant LED value arrays on the high level.

kareltucek, 2023-12-18:
I don't think that is true. Monochrome handling of UHK v1 and RGB handling of key cluster seem to me as two quite significantly different issues.

mhantsch, 2023-12-18:
@kareltucek I should have phrased my statement as a question: "Isn't this issue almost the same....?", and you already provided a good answer as to why you want to handle them are two separate issues; thanks.

PR #739 (2024-01-05): Implement backlight for v1.
Closes #569.
Closes #655.
closed 2024-01-05 by mondalaci