UltimateHackingKeyboard/firmware #570: OR/AND gates for ifKeyActive command

rftw, 2022-12-04:
This is a feature request.

I wish to pass more than one keyId to the ifKeyActive command and check if one of them is active (OR condition) or all of them are active (AND condition)

Currently for as a workaround for OR condition, I use each keyId on a separate line:
```
ifKeyActive 10 final ...
ifKeyActive 11 final ...
...
```
However I couldn't find a way to check if two keys are active at the same time.

kareltucek, 2022-12-04:
> However I couldn't find a way to check if two keys are active at the same time.

`ifKeyActive 10 ifKeyActive 11 final ...`

rftw, 2022-12-04:
Thank you! 
I initially attempted to pass multiple keyids `ifKeyActive 10 11 ...` assuming the same syntax as `ifShortcut` command.

closed 2022-12-04 by rftw