UltimateHackingKeyboard/firmware #571: scroll navigationMode doesn't work in control layer

supermomonga, 2022-12-05:
```
set module.touchpad.navigationMode.control scroll
```

It doesn't work with both Trackball/Touchpad modules.

If I change `{layerId}` or `{navigationMode}` to other value, it works as expected.

for example, following settings does work.

```
set module.touchpad.navigationMode.control zoom
set module.touchpad.navigationMode.control zoom
set module.touchpad.navigationMode.mod zoom
```

----

>  Right keyboard half firmware: UltimateHackingKeyboard/firmware:9.1.4
 Left keyboard half firmware: UltimateHackingKeyboard/firmware:9.1.4
 Key cluster firmware: UltimateHackingKeyboard/firmware:9.1.4
 Touchpad is detected, and its firmware is implemented by the right keyboard half.

supermomonga, 2022-12-05:
OK, I realized that maybe it triggers both `Ctrl key down` and `scrolling`, so it triggers unexpected behavior. (In Windows, Ctrl+scroll triggers zoom in/out.)

I think this also will be happened with alt/shift/super.
Can I configure to simply trigger only `{navigationMode}`'s keycode, without `{layerId}`'s keycode?

kareltucek, 2022-12-05:
No.

supermomonga, 2022-12-05:
Is there an any plan to make it configurable?

kareltucek, 2022-12-05:
Not at the moment. Maybe if there is a significant demand for it, or if there is a good usecase for it.

Also, what is your usecase? Why do not regular layers do the job?

Some random thoughts:

- I don't think the requested behaviour should be made default. Not unless fallthrough logic is applied to navigation modes as well (i.e., setting `navigationMode.control none` would fall through to base layer), because then it would break ctrl + mouse usecases in configurations where both layers are set to `cursor`.

- I think that coupling navigation modes to modifiers is an antipattern, it limits the set of reachable keyboard commands. (In other words, it makes it impossible to zoom in/out.)

- It is also a technical challenge, because since ctrl and mouse events are reported through two different USB interfaces, some delay between those has to be ensured. (Otherwise, reports will not always arrive in correct order, and host OS will interpret them sometimes with control, sometimes without.) This is a problem because navigation modes are exempt from postponing logic (for obvious reasons).

- This also relates to the philosophical question of having navigation modes coupled to layers.

- I think this is quite an edge case. It would be hard to communicate such a configuration does, and thus it would mostly just clutter the syntax and documentation.

(Closing, but feel free to follow up.)

closed 2022-12-05 by kareltucek

kareltucek, 2022-12-05:
> @kareltucek Are navigation modes expected to compose with the modifiers of modifier layers? I think the expected behavior is not to compose. 

I think not. But we do not have any formal specification...

> Maybe we should disallow using navigation modes via modifier layers.

I don't have a strong opinion on this. 

Generally, I lean towards not limiting the feature set artificially, but indeed here it would set a clear boundary of what the intended usecases are.

My strongest feeling about this is that disallowing it will either make the grammar partially invalid, or need a new rule for non-modifier layers. (Bot are undesirable.)

mondalaci, 2022-12-05:
Maybe we should wait and see how users are affected by navigation modes on modifier layers. Depending on their level of confusion, we could make navigation modes inaccessible via modifier layers.

kareltucek, 2022-12-05:
Ok!

supermomonga, 2022-12-06:
Thank you for the detailed explanation.

I understood that this is an edge case and that it's difficult to make it configurable from a technical view or complexity, as you explained.

I think maybe some users will get confused like me, so once the official specifications are decided, I would appreciate it if you could specify them in the Agent's built-in user manual.

----

My usecase:

I put the LCtrl key in the most accessible position from my finger. This is because I use the shortcut keys combined with LCtrl the most frequently. This includes many of the shortcut keys that the OS provides by default, such as C-c, C-x, and C-v.

I use scrolling very frequently(web-browsing, reading books, etc...) and I thought it would be very useful to be able to trigger touchpad/trackball scrolling with the this most accessible key(LCtrl).

```
set module.touchpad.navigationMode.control scroll
set module.trackball.navigationMode.control scroll
```