UltimateHackingKeyboard/firmware #575: Allow any key in ifShortcut

likern, 2022-12-11:
It would be awesome to support range of keys or `any` keyword in `ifShortcut`.

My command is the following
```
ifShortcut timeoutIn 200 29 exec toMouseLayer
// Here I want not only 16 key, but all keys in layer fn4
ifShortcut noConsume timeoutIn 1000 16 final holdLayer fn4 
autoRepeat tapKey space 
```

This (`ifShortcut noConsume timeoutIn 1000 16 final holdLayer fn4 `) is the only what works the way I like and reliably.
So, when I press `space + <some key>` I get key from layer `fn4`, while just space (hitting fast multiple times) and autorepeat space still works.

**Note:** Looking at source code I've found that `%0` is similar to what I wanted
```
ifShortcut noConsume timeoutIn 1000 %0 final holdLayer fn4
```
it produces desired key (at `fn4` layer), while later giving me error
```
spACeaNDTOmOuSe; 0; 1; nOT enough pending keys1 note tHAT this is zero-indexed1
```

mondalaci, 2022-12-11:
It seems to me that the issue has been (partially?) resolved. I'm closing this issue, but feel free to follow up.

likern, 2023-01-09:
@mondalaci Can we reopen this (I can't)?
@kareltucek

I want this behaviour and think `ifSortcut` might help.

1. Press key `KEY`
2. Wait for some time gap (350ms), do nothing.
3. **After gap** if `KEY` is still pressing and no other keys were pressed during gap (no interruptions) - press `backspace` in sticky mode (autorepeat)
4. **During gap** if `KEY` was released and no other keys were pressed before release (no interruptions) - tap `backspace` once. This happens as soon as possible, so fast taps are supported.
5. **During gap** if `KEY` was interrupted by some key `OTHER_KEY` it works as `Ctrl`, tap `Ctrl + OTHER_KEY`. If `KEY` is still pressing continue pressing `Ctrl` - so `KEY + A + B` works as `Ctrl-A Ctrl-B`.

So this behaviour essentially fits into category Primary/Secondary roles. But now Primary/Secondary doesn't support this use case (at least I couldn't make it working), only cases where primary role holds modifiers which are invisible if released.

As far as I understand this should help:
```
ifShortcut noConsume timeoutIn 350 any_key final tapKeySeq C-any_key
holdKey backspace
```

mondalaci, 2023-01-12:
@kareltucek Can you comment on this?

kareltucek, 2023-01-12:
Using ifshortcut as primary/secondary role is an abuse of the command. 

But indeed implementation may be enough simple so that there might not be sufficient reason not to do it.

likern, 2023-01-12:
> Using ifshortcut as primary/secondary role is an abuse of the command.
> 
> But indeed implementation may be enough simple so that there might not be sufficient reason not to do it.

I'm open to better and more robust solution. Updating `ifShortcut` is something came to my mind. I agree the desing is kind of ugly. If we can we need something better.

kareltucek, 2023-07-07:
Since advanced secondary roles have been implemented, I would rather close this.

closed 2023-07-07 by kareltucek