UltimateHackingKeyboard/firmware #576: Backlight turns back on after computer went to sleep

sebastiankr, 2022-12-13:
Computer does not wake up, but the keyboard LED and backlight turn back on after a while.
Happens on USB-A dongle as well as direct USB-C. With or without the Key Cluster. S3 save mode is enabled.

Windows 10, Motherboard Z690 UD AX DDR4, Intel Alder Lake
[UserConfiguration.zip](https://github.com/UltimateHackingKeyboard/firmware/files/10213883/UserConfiguration.zip)

mondalaci, 2022-12-13:
Are you running the latest 9.1.4 firmware? If so, is there a specific time interval after which the backlight turns back on?

mondalaci, 2022-12-15:
@sebastiankr This is probably a newly introduced bug. Can you please revert to earlier firmware versions one by one (9.1.4 -> 9.1.3 -> ...) and see which one is affected?

sebastiankr, 2022-12-17:
I tested with 9.0.1. Issue still persists.

mondalaci, 2022-12-18:
Thanks for testing! Please test with earlier firmware releases all the way back to 8.10.12

sebastiankr, 2022-12-18:
With 9.0.1, the light turned on, even when the PC was shutdown. It seemed to be at the same time the AC of the house kicked in.

I am not able to install anything earlier.  Flashing fails on the left keyboard side. 

Start flashing firmware
[DeviceService] Operating system: Windows_NT 10.0.19045 x64
[DeviceService] Agent version: 2.0.2
[DeviceService] New firmware version: 8.10.12
[DeviceService] New firmware user config version: 4.2.0
[DeviceService] stop poll UHK Device
[DeviceService] stopped poll UHK Device
[DeviceOperation] Read "leftHalf" version information
[DeviceOperation] Read "leftHalf" repo information
[DeviceService] UHK Device firmware upgrade starts: {"id":2,"name":"UHK 60 v2","vendorId":"0x1D50","keyboardPid":"0x6124","bootloaderPid":"0x6123","buspalPid":"0x6121"}
[DeviceService] Device right firmware version: 9.0.1
[UhkOperations] Start flashing right firmware
[UhkOperations] Reenumerate bootloader
[UhkHidDevice] Start reenumeration, mode: Bootloader, timeout: 5000ms
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting....
[UhkHidDevice] Available devices changed.
[UhkHidDevice] Removed: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_03#7&361d9ceb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":3,"usagePage":1,"usage":128}
[UhkHidDevice] Removed: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_02#7&2ed59145&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":2,"usagePage":12,"usage":1}
[UhkHidDevice] Removed: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_00#7&22c1dfc8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":0,"usagePage":65280,"usage":1}
[UhkHidDevice] Removed: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_01#7&afe5383&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":1,"usagePage":1,"usage":6}
[UhkHidDevice] Removed: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_04#7&12465f29&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":4,"usagePage":1,"usage":2}
[UhkHidDevice] Could not find reenumerated device: Bootloader. Waiting.....
[UhkHidDevice] Reenumerating devices
[UhkHidDevice] Device communication closing.
[UhkHidDevice] Device communication closed.
[UhkOperations] Waiting for bootloader
[UhkOperations] Flash security disable
[UhkOperations] Flash erase region
[UhkOperations] Read RIGHT firmware from file
[UhkOperations] Write memory
[UhkOperations] Reset bootloader
[UhkOperations] Close communication channels
[UhkOperations] Right firmware successfully flashed
[DeviceService] Waiting for keyboard
[UhkHidDevice] Available devices changed.
[UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_03#7&361d9ceb&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":3,"usagePage":1,"usage":128}
[UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_02#7&2ed59145&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":2,"usagePage":12,"usage":1}
[UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_00#7&22c1dfc8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":0,"usagePage":65280,"usage":1}
[UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_01#7&afe5383&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":1,"usagePage":1,"usage":6}
[UhkHidDevice] Added: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_04#7&12465f29&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":4,"usagePage":1,"usage":2}
[UhkHidDevice] Used device: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_00#7&22c1dfc8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":0,"usagePage":65280,"usage":1}
[DeviceOperation] Transferring error Error: Communications error with UHK. Response code: 3
    at C:\Users\skropp\AppData\Local\Programs\uhk-agent\resources\app.asar\electron-main.js:2:620262
[UhkHidDevice] Device communication closing.
[UhkHidDevice] Device communication closed.
[DeviceService] updateFirmware error {"message":"Communications error with UHK. Response code: 3","stack":"Error: Communications error with UHK. Response code: 3\n    at C:\\Users\\skropp\\AppData\\Local\\Programs\\uhk-agent\\resources\\app.asar\\electron-main.js:2:620262"}
[UhkHidDevice] Available devices unchanged
[UhkHidDevice] Used device: {"vendorId":"0x1D50","productId":"0x6124","path":"\\\\?\\hid#vid_1d50&pid_6124&mi_00#7&22c1dfc8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","manufacturer":"Ultimate Gadget Laboratories","product":"UHK 60 v2","release":258,"interface":0,"usagePage":65280,"usage":1}
[DeviceOperation] Read "leftHalf" version information
[DeviceService] start poll UHK Device
Communications error with UHK. Response code: 3

mondalaci, 2022-12-18:
@ert78gb Can you help?

mondalaci, 2022-12-18:
@kareltucek Can you please look into error code 3?

ert78gb, 2022-12-18:
Some info. When agent tries to save the user config to the keyboard then the firmware returns with 3 response code.

My question is how could we map the response codes to response messages?

kareltucek, 2022-12-18:
> Can you please look into error code 3?

```
    typedef enum {
        ParserError_Success                             =  0,
        ParserError_InvalidSerializedKeystrokeType      =  1,
        ParserError_InvalidSerializedMouseAction        =  2,
        ParserError_InvalidSerializedKeyActionType      =  3,
        ParserError_InvalidLayerCount                   =  4,
        ParserError_InvalidModuleCount                  =  5,
        ParserError_InvalidActionCount                  =  6,
        ParserError_InvalidSerializedMacroActionType    =  7,
        ParserError_InvalidSerializedSwitchKeymapAction =  8,
        ParserError_InvalidModuleConfigurationCount     =  9,
        ParserError_InvalidKeymapCount                  = 10,
        ParserError_InvalidAbbreviationLen              = 11,
        ParserError_InvalidMacroCount                   = 12,
        ParserError_InvalidSerializedPlayMacroAction    = 13,
        ParserError_InvalidMouseKineticProperty         = 14,
        ParserError_InvalidLayerId                      = 15,
    } parser_error_t;
```

---

@sebastiankr in order to downgrade to 8.x.x firmware, you will need to use a compatible agent. (e.g., [1.5.17](https://github.com/UltimateHackingKeyboard/agent/releases/tag/v1.5.17)) and reset the user config from there.

kareltucek, 2022-12-18:
@mondalaci I guess this is simply caused by the change in config format - the one where we went from static layers to dynamic layers.

When downgrading the firmware, new config still remains... consequently, layer counts/ids are interpreted as actions... action counts get mismatched... and sooner or later, an "invalid" "action" is encountered, because by that time the parser is reading randomly offset ("random") data.

I wonder whether the firmware should check UserConfig version and complain when incompatible. Not sure how that should be done. I guess the version range should be hardcoded in the code, checking major and minor semver version?

kareltucek, 2022-12-18:
Also, I find it a bit confusing that Agent states just `Communications error with UHK. Response code: 3` without saying what type of error it is. 

That is to say that I am just guessing that the error is a config parser error. (If I am correct, it is being returned returned as byte 0 in response to ApplyConfig USB command.

ert78gb, 2022-12-18:
The error occurred, when saved the user config. I don't log the "transfer" and "apply" communication on default log level, but I will extend the log messages with these info too.

@kareltucek  The above mentioned error codes are the ApplyConfig error codes?

kareltucek, 2022-12-18:
> The above mentioned error codes are the ApplyConfig error codes?

Yes.

> The error occurred, when saved the user config. I don't log the "transfer" and "apply" communication on default log level, but I will extend the log messages with these info too.

So the 3 should be from UsbCommandId_WriteStagingUserConfig command?

If so, then this should be relevant:

```
    typedef enum {
        UsbStatusCode_WriteConfig_LengthTooLarge    = 2,
        UsbStatusCode_WriteConfig_BufferOutOfBounds = 3,
    } usb_status_code_write_config_t;
```

I don't see how such a problem should be related to firmware downgrades though.

Should I investigate?

ert78gb, 2022-12-18:
I don't think you have to investigate, because if the user config larger than 32kb then Agent shows an error message to the user.

kareltucek, 2022-12-18:
Well, that is clearly not what's happening in the scenario above. That user config has 4kb.

ert78gb, 2022-12-18:
I meant the problem is not the `UsbStatusCode_WriteConfig_BufferOutOfBounds`. Sorry I was not clear

mondalaci, 2022-12-19:
> @sebastiankr in order to downgrade to 8.x.x firmware, you will need to use a compatible agent. (e.g., [1.5.17](https://github.com/UltimateHackingKeyboard/agent/releases/tag/v1.5.17)) and reset the user config from there.

@sebastiankr Please downgrade according to the above instruction. Sorry for the trouble, and thank you for your perseverance!

> I wonder whether the firmware should check UserConfig version and complain when incompatible. Not sure how that should be done. I guess the version range should be hardcoded in the code, checking major and minor semver version?

@kareltucek These were also my thoughts, but if the firmware cannot parse the user config, the LED display will show FTY. The root of the problem is that Agent tries to save the user config after firmware updates which is useful when upgrading the user config version but counterproductive when downgrading as Agent can't downgrade the user config. (We could implement user config downgrade, but it's not worth the effort.) We'll solve the issue by making Agent not save the user config after firmware downgrades.

sebastiankr, 2022-12-20:
I connected the computer to a surge protector socket. This seems to have fixed the issue. Testing this takes quite some time, because the issue was sporadic and unpredictable.

Happy to test lower versions after.

mondalaci, 2022-12-20:
@sebastiankr Thanks for your feedback! Please keep us updated.

mondalaci, 2023-09-11:
Since we've received no further updates or related bug reports, I consider this issue resolved.

closed 2023-09-11 by mondalaci