UltimateHackingKeyboard/firmware #577: how to create macro with holding space + left mouse

msio, 2022-12-15:
I'm trying to create macro with holding space + left mouse. I want to achieve the same behaviour as you would hold left mouse and space. This kind of behaviour is mostly used in Graphics program when you want to move whole canvas.  
How can I achieve that by macros please

Thanks for any help
Cheers

mondalaci, 2022-12-15:
@kareltucek Any ideas?

kareltucek, 2022-12-15:
The trick is the `delayUntilRelease` macro command.

kareltucek, 2022-12-15:
Also, you may need to enable the extended macro command mode, which can be done by creating a macro named `$onInit` that contains (at least) `set macroEngine.extendedCommands 1`.

likern, 2023-01-19:
I think this should work
```
holdKey space
holdKey mouseBtnLeft
```
it doesn't (tested in Figma). Looks like a bug to me.

likern, 2023-01-19:
> 

```
pressKey space
pressKey mouseBtnLeft
delayUntilRelease
```

kareltucek, 2023-06-19:
I believe this is solved? Feel free to protest!

closed 2023-06-19 by kareltucek