UltimateHackingKeyboard/firmware #581: Refactor macros.c

mondalaci, 2022-12-28:
`macros.c` contains a whopping 3466 lines, which should be refactored in the near future.

Luckily, the content of the file seems well structured. I propose creating a `macros` directory, and using individual files per macro command, similarly to the `usb_commands` directory.

likern, 2023-01-09:
Have you considered to use `Zig` language? It has awesome interoperability with plain C (you can just compile C with Zig compiler), while it's also much safer and many things are just built-in. This can help to catch many hidden bugs and improve firmware reliability in general.

mondalaci, 2023-01-09:
@likern Thanks for the idea, but I don't plan to introduce another language.

likern, 2023-01-09:
Yep, understandable. But I had to try ;))

mondalaci, 2023-09-06:
I'm closing this issue because we could have a thousand refactoring issues like this. Eventually, we'll get to them.

But as a last word, I'd like to note that it'd be worth using gpref at some places, which would make braching faster, such as for macro commands.

closed 2023-09-06 by mondalaci

kareltucek, 2023-09-06:
I was thinking about a similar refactor, although I intended to use a normal lets-reinvent-the-wheel super-naive hash table.

I am also thinking about compiling the macros into a sort of bytecode. Compilation would take place right after user config is loaded, and the compiled bytecode would be placed in the unused validation buffer.

(It would make things much simpler for the promised scoping, as data about scope begins/ends could be placed right in the bytecode, just as label jump adresses.)

mondalaci, 2023-09-06:
gperf creates a collision-free hash, which is worth considering, but alternative solutions, like the ones suggested, may work just as well. I trust your judgment and am glad you're thinking about this.