UltimateHackingKeyboard/firmware #582: Minus key doesn't work with ifShift

likern, 2023-01-09:
**Doesn't work**
```
ifShift suppressMods tapKey -
ifNotShift tapKey _
```
**Works**
```
ifShift suppressMods write -
ifNotShift write _
```
while other keys (0-9) are working with `tapKey`.

I'm using this feature to make keys reversible.

[image]
[image]
[image]

mondalaci, 2023-01-12:
@kareltucek Can you reproduce the issue?

kareltucek, 2023-01-12:
Yes, will fix it soon...

PR #583 (2023-01-14): Fix shortcut parser to parse "-" correctly.
Closes #582.
closed 2023-01-14 by mondalaci