UltimateHackingKeyboard/firmware #586: module.trackpoint.navigationMode.shift doesn't apply changes

nhance, 2023-01-19:


kareltucek, 2023-01-19:
Agreed.

mondalaci, 2023-01-19:
@kareltucek Then please make it happen as time allows.

@nhance I am sorry for axing this feature, but it's too problematic.

nhance, 2023-01-19:
Understood. Does this mean the existing functionality for MOD + Trackpoint to scroll will disappear?

I was hoping to have the ability to use the trackpoint for mouse and scroll using only a single half of the keyboard.

mondalaci, 2023-01-19:
Navigation modes are very useful, so they're here to stay. We'll only disallow them via modifier layers.

nhance, 2023-01-19:
Understood. I will miss the comfortable option of control or shift plus trackpoint for scrolling. Not many other keys are within comfortable reach.

mondalaci, 2023-01-19:
Reopening because we have yet to disallow modifier layers for navigation modes.

PR #617 (2023-06-19): Disallow navigation modes in modifier layers.
Closes #586
closed 2023-06-19 by mondalaci