UltimateHackingKeyboard/firmware #587: ifGesture strange behaviour

likern, 2023-01-20:
I have the following macro, which works kind of like "One-Shot Key" and "Two-Shot Key" for left space.

```
# Right side top row
ifGesture 14 final holdKey *
ifGesture 7 final holdKey =
ifGesture 8 final holdKey "
ifGesture 9 final holdKey ;
ifGesture 10 final holdKey '

ifGesture 81 14 final holdKey +
ifGesture 81 7 final holdKey @
ifGesture 81 8 final holdKey ?
ifGesture 81 9 final holdKey &
ifGesture 10 final holdKey '

# Right side middle row
ifGesture 21 final holdKey _
ifGesture 15 final holdKey : 
ifGesture 16 final holdKey .
ifGesture 17 final holdKey ,

ifGesture 81 21 final holdKey -
ifGesture 81 15 final holdKey / 
ifGesture 81 16 final holdKey #
ifGesture 81 17 final holdKey $

# Right side bottom row
ifGesture 22 final holdKey (
ifGesture 23 final holdKey {
ifGesture 24 final holdKey [
ifGesture 25 final holdKey <

ifGesture 81 22 final holdKey )
ifGesture 81 23 final holdKey }
ifGesture 81 24 final holdKey ]
ifGesture 81 25 final holdKey >
```

I've experienced the strange behaviour. If I press normal key `e` (for example) and while holding it I press `left space` (which activates macro) and then release both keys - pressing `e` continues on the screen for some time.

While if I just holding key `e` and release it, pressing stops immediately.

**How To Reproduce:**
1.  Press key `e` and then (while holding it) press `left space` (to which macro was assigned)

[Screencast from 2023-01-20 13-36-27.webm](https://user-images.githubusercontent.com/842306/213677651-db906a1b-0fcb-4e24-8662-3d40f2cc0e3d.webm)

likern, 2023-01-20:
Better reproduction. Here I press key `a`, then `left space` (macro activator) and release immediately.
Key `a` release event might interleave with `space` press event (if tapping keys fast enough), but not sure.

[Screencast from 2023-01-20 15-36-03.webm](https://user-images.githubusercontent.com/842306/213696485-c8556658-b108-421f-a7ec-aaa56c9a7038.webm)

kareltucek, 2023-01-20:
Yes, ifGesture has to query future keystrokes in order to decide whether to act. If it decides _not_ to act, UHK is required to behave as if the condition did not take place at all... ...and preserve the order of all presses _and releases_. 

(I guess this is another reason why ifGesture is different from one-shot keys. One-shot keys don't need to mess with time in order to do stuff.)

I.e., at the moment this is known and is by design. I guess we may decide to change that policy and let releases take place before their proper time in the future though.

likern, 2023-01-21:
> Yes, ifGesture has to query future keystrokes in order to decide whether to act. If it decides _not_ to act, UHK is required to behave as if the condition did not take place at all... ...and preserve the order of all presses _and releases_.
> 
> (I guess this is another reason why ifGesture is different from one-shot keys. One-shot keys don't need to mess with time in order to do stuff.)
> 
> I.e., at the moment this is known and is by design. I guess we may decide to change that policy in the future and let releases take place before their proper time.

But how does it relate to autorepeating `a` key, while I'm not 🚫 holding it?

kareltucek, 2023-01-21:
Well, it means that if a key is released *after* an `ifGesture` is triggered, we aren't allowed to release it until we know the outcome of the ifGesture, because any action that is made depending on that `ifGesture` has to come before that key release.

(The actual autorepeat is your OS feature, it has nothing to do with UHK.)

likern, 2023-01-21:
OS autorepeat stops producing `a` key as soon as I release `a` key.

While in mentioned issue it produces about 30 `a` after I released `a` key.

I press `a`, press `left space`, release both keys, so I do nothing and see 30 more `a` typing.

likern, 2023-01-21:
> Well, it means that if a key is released _after_ an `ifGesture` is triggered, we aren't allowed to release it until we know the outcome of the ifGesture, because any action that is made depending on [...]

Got it, probably. So that means that `a` keeps pressing all that max timeout of `ifGesture`? Am I correct?

I think `tapKey` should solve this problem?
But how to make it work with autorepeat so I could produce `======` or `************` easily?

kareltucek, 2023-01-21:
> . So that means that `a` keeps pressing all that max timeout of `ifGesture`? Am I correct?

Correct.

> I think `tapKey` should solve this problem? But how to make it work with autorepeat so I could produce `======` or `************` easily?

Maybe just increase the OS autorepeat delay? Or figure out a setup that does not need to overlap ifGesture with other keys?

kareltucek, 2023-06-19:
I think we can close this?

closed 2023-06-19 by kareltucek