UltimateHackingKeyboard/firmware #588: show active layer on 14-segment display

mhantsch, 2023-01-25:
At the moment, the display shows the active keymap, and layers are indicated with the "mod", "fn" and "mouse" LEDs.

Now that the firmware also has "fn2-fn5" (as well as "shift/alt/ctrl/super") layers, there are no indicators for there layers.

I would like to be able to show different text on the segment display when layers are switched. My use case: the secondary role of my Tab key switches to a layer with a num pad. I would like to display "NUM" when that layer is active.

mondalaci, 2023-01-25:
Have you looked into smart macros? Maybe it's possible to implement your use case with them.

kareltucek, 2023-01-25:
@mondalaci we don't have `$onLayerChange` event (at the moment anyway).

But yes, I find implementing `$onLayerChange` event to be a relatively reasonable request, which can be done quite easily.

mondalaci, 2023-01-25:
@kareltucek Sounds good to me. How exactly would one implement @mhantsch's use case with the help of `$onLayerChange`?

kareltucek, 2023-01-25:
Have one macro for each layer, e.g.:

`$onLayerChange fn`
```
setLedTxt 0 fn
```

mondalaci, 2023-01-25:
Makes perfect sense. Feel free to implement `$onLayerChange`.

mondalaci, 2023-06-21:
Sorry, moving my comment to the PR.

closed 2023-06-21 by mondalaci

mhantsch, 2023-07-04:
What's the syntax for the onLayerChange and onKeymapLayerChange macro events? `$onKeymapLayerChange QWR mouse` ? `$onLayerChange fn2` ?

kareltucek, 2023-07-04:
Yes. Namely:

```
    $onLayerChange {LAYERID|any}
    $onKeymapLayerChange KEYMAPID LAYERID
```

Docs are fixed now, thanks for pointing out!

PR #621 (2023-06-21): Add onLayerChange and onKeymapLayerChange macro events.
Closes #588