UltimateHackingKeyboard/firmware #589: Feature Request: Independently control key cluster mini trackball axis

shayded-exe, 2023-02-06:
After swapping the axis of the mini trackball, right scrolls down and up scrolls right.

I want to avoid accidentally horizontally scrolling to the right when trying to scroll down. When moving the trackball right, I often accidentally move it up as well. I've adjusted the axis locking settings, but I still find it too easy to accidentally scroll horizontally.

I propose two additions:
1. The ability to individually disable axis (i.e. only allow vertical scrolling).
2. The ability to independently invert axis

mondalaci, 2023-02-07:
I think it's already possible to configure what you're after.

See the "Macro name example" section of "set » navigation mode action". The relevant macros should contain one mouse scroll action each.

closed 2023-02-07 by mondalaci

shayded-exe, 2023-02-08:
I see, thanks!

The only issue is that it requires using up one of the other navigation modes, but this isn't really a problem for me since I don't need the zoomMac mode.