UltimateHackingKeyboard/firmware #590: Trackball accuracy

Vistaus, 2023-03-03:
I apologize in advance for maybe not asking the question in the right place, but I didn't know where else to ask.

I still love my UHKv2 and I may be interested in buying the trackball module. In the shop there's a video demo done on Linux Mint and the trackball seems to be super accurate! My current trackball is a Logitech MX Ergo and is really good, but the accuracy is less good then what I saw in your video (I'm also on Linux). How did you manage to adjust the firmware in such a way to make the trackball so accurate (under Linux)?

closed 2023-03-03 by mondalaci

Vistaus, 2023-03-03:
That's not what I asked…

mondalaci, 2023-03-03:
I've read your question twice, and I'd still give the same answer. Please clarify your question, and I'll try my best to give you a proper answer.

Vistaus, 2023-03-03:
I don't want to know how to adjust the module speed, I want to know how exactly you achieved with the firmware to make the trackball so **accurate** under Linux.

mondalaci, 2023-03-03:
I may have actually tweaked the module speed via smart macros. I don't remember.

But non-UHK input devices have nothing to do with the UHK firmware, and they have everything to do with their DPI setting and the mouse sensitivity settings of the OS.

Vistaus, 2023-03-03:
> I may have actually tweaked the module speed via smart macros. I don't remember.

Too bad you don't remember. I want to achieve the same results with my current trackball, but also with the trackball module once I order it.

> But non-UHK input devices have nothing to do with the UHK firmware, and they have everything to do with their DPI setting and the mouse sensitivity settings of the OS.

I know. But my question was not just about my current trackball, but also about the trackball module as I may be interested in buying it.

mondalaci, 2023-03-03:
I read this thread again and still think I provided correct answers, but maybe we're crossing wires.

Even though I don't remember whether I tweaked the default settings of the trackball module at the time of making the video, as I said, its speed settings can be adjusted via smart macros regardless. The following screenshot might provide a better explanation.

[image]

I have the trackpoint module attached, but the same settings apply to the trackball module.

The speed and acceleration are the functions of device settings and OS settings. For non-UHK devices, the device settings can be configured with device-specific applications or switches.

kareltucek, 2023-03-04:
> How did you manage to adjust the firmware in such a way to make the trackball so accurate (under Linux)?

We implement an acceleration driver in UHK.

For human-friendly explanation, see the agent smart macro reference (or screen above), for more technical explanation search https://github.com/UltimateHackingKeyboard/firmware/blob/master/doc-dev/reference-manual.md for `xceleration` .

Vistaus, 2023-03-07:
Ah, thank you both for the explanation! 🙂