UltimateHackingKeyboard/firmware #592: putting two sequential write commands into a macro causes the first character of the second write to not show up

opensensesolutions, 2023-03-15:
Here's an odd issue that isn't really a problem in practice but I thought I'd report it since it could have larger implications once the root cause is discovered.

If you create a macro with a "Type text" action to write the text "test7a." , and then add a second "Type text" action to write the text "test7b." and trigger the macro you get:

test7a.est7b.

Note that the second t is missing.  If you put newlines at the end of each write text block instead of periods you get:
test7a
est7b

If you do a third write immediately after the second it's first character is also missing.

Putting anything in between the write commands eliminates the problem, such as a delay of 1 ms or even a Tap key LShift.

The same thing occurs with smart macros using write/delayUntil/tapKey commands.

It's not hard to put a dummy operation in between, and most of the time you could just combine the two writes into one command, so I'm certainly not waiting on a fix. 

I am using UHK.Agent-2.1.1-linux-x86_64.AppImage on ubuntu 22.04 and updated the firmware to 9.2.1 ( bundled with agent) . I delete the .config/uhk-agent directory after every uhk-agent run to avoid storing my password macros in plain text so there shouldn't be any old cruft affecting things.  This occurred through several device power cycles and agent writes.

mondalaci, 2023-03-15:
@opensensesolutions Thanks so much for reporting!

@kareltucek Would you fix this bug?

kareltucek, 2023-03-15:
Yes, will look into it.

PR #616 (2023-06-19): Send empty report after every write command.
Otherwise, consider words 'first', 'second' as two write actions. If there is no empty report inbetween, output is 'firstecond'.

Closes #592.
closed 2023-06-19 by mondalaci