UltimateHackingKeyboard/firmware #598: resolveSecondary doesn't seem to work properly

shayded-exe, 2023-04-01:
I've read the docs multiple times and looked over the source and I can't figure out why this isn't working properly.

I have the macro below bound to my `s` key.
The 1500ms timeout is to make it easier to illustrate the problem.
```
resolveSecondary 1500 50 1 3
tapKey s
break
holdLayer fn
```

From the [reference manual](https://github.com/UltimateHackingKeyboard/firmware/blob/master/doc-dev/reference-manual.md):
```
In more detail, the resolution waits for the first key release - if the switch key is released first or within the safety margin delay after release of the postponed key, it is taken for a primary action and goes to the section of the "primary action", then the postponed key is activated; if the postponed key is released first, then the switcher branches the secondary role (e.g., activates layer hold) and then the postponed key is activated; if the time given by first argument passes, the "secondary" branch is activated as in the previous case.
```

This makes it clear that assuming the safety margin is respected, that releasing the postponed key before the trigger key will trigger the secondary action regardless of if the first timeout has been reached.

**Test**:
1. Press and hold `s`
2. Tap `j` (which is bound to a symbol on my fn layer)
3. Release `s` after another 50ms but still before the 1500ms timeout.

**Expected Result**:
The symbol on the fn layer will be typed

**Actual Result**:
`sj` is typed

The only way I can get the secondary action to activate, is if I hold the trigger key for the entire timeout (1500ms in this case).

Another strange result I observed is that if you set the safety margin to 0ms, the secondary action activates as expected, but it's impossible to activate the primary action.

I examined the source and I noticed something that seemed odd (maybe I just don't understand it).

https://github.com/UltimateHackingKeyboard/firmware/blob/115a6a4f54acadea5804e029be41ae5df6190fbe/right/src/macros.c#L1720-L1748

Specifically line 1723:
```c
bool pendingReleased = PostponerQuery_IsKeyReleased(s->ms.currentMacroKey);
```

`currentMacroKey` is assigned to the key that started the macro, is it not? The way the algorithm is written seems to expect that `pendingReleased` is whether or not the second key has been released (`j` in my case), but `currentMacroKey` is only ever assigned to the macro trigger key and I can't find anywhere it's ever assigned to the key that was pressed since the macro started.

Regardless, I have been unable to get `resolveSecondary` to produce the behavior that the docs suggest.

kareltucek, 2023-04-02:
You are correct!

It looks like a typo made during a postponer refactor. I believe the line should be `bool pendingReleased = PostponerExtended_IsPendingKeyReleased(0);`

shayded-exe, 2023-04-02:
Awesome, thank you!

PR #601 (2023-04-03): Fix logic of resolveSecondary.
Closes #598.
closed 2023-04-03 by mondalaci