UltimateHackingKeyboard/firmware #607: Smart Macro: Single Word Caps Lock

mstate, 2023-04-19:
I was hoping to get some advice on how to create a smart macro that turns on caps lock until I press the space bar.  This would let me caps lock a single word, especially useful for constants.

For reference, the Moonlander keyboard has this functionality out of the box.

UndeadKernel, 2023-04-20:
The macros user guide has exactly this example already there.
In the file `doc-dev/user-guide.md` search for the words `caps words`. You will find the answers you are looking for somewhere around the search hit.

mondalaci, 2023-04-21:
@kareltucek Does the stated "workaround" have any practical deficiencies?

kareltucek, 2023-04-22:
I guess the proper implementation should deactivate the caps lock by any non-letter scancode. 

The (suggested) workaround deactivates the caps lock by space only. (Or more generally, you have to make a separate macro for every key that is supposed to deactivate the caps lock, which is pain.)

I agree that it is not a priority, but it would be nice to have some way to specify generic trigger conditions which would be able to cover a wider range of similar use-cases...

mondalaci, 2023-04-22:
If anyone is reading this, feel free to elaborate on the desired trigger condition so that we can understand your needs better.

kareltucek, 2023-08-22:
Note to self: Usecase: prefix keytaps with an action. This could use an ability to wait until postponer processes given number of keys. Or maybe this could be implemented by generalizing ifInterrupted as an uint8_t.