UltimateHackingKeyboard/firmware #608: control layer: inconsistent ctrl modifier engaged

UndeadKernel, 2023-05-04:
There seems to be an inconsistency in how the ctrl modifier is treated in the control layer. I would expect the next two scenarios to behave the same:
1) If I assign (e.g.) `right arrow` to a key in the ctrl layer, pressing the assigned key moves one character at a time (i.e., the control modifier **is not engaged**, IMHO, as desired).
2) If I assign a macro with only the line `holdKey rightArrow`, pressing the assigned key moves one word at a time (i.e., the control modifier **IS engaged**).

Is this behavior expected? Shouldn't the control modified not be engaged in both scenarios?

UndeadKernel, 2023-05-04:
In case this is relevant to figure out what the desired behavior should be: The second scenario (see the original post) can be achieved without engaging the ctrl modifier with:
`supressMods holdKey rightArrow`.

kareltucek, 2023-05-04:
Noted. I tentatively agree that ctrl probably should not be engaged. Not sure at the moment what (and whether anything) it means for the general logic of sticky modifiers. (Modifier layers are implemented as a negative sticky mod mask.)

Also, there have been some tickets regarding similar issue with mouse behavior, which however I did not consider relevant, at least at the time.

kareltucek, 2023-07-17:
To make specification clear...

@mondalaci in your opinion:

- Should non-keypress actions in a modifier layer automatically receive the modifier?
- Should a navigation mode inherited from base layer receive the modifier?

kareltucek, 2023-07-17:
Nevermind, I think I have an answer.

mondalaci, 2023-07-17:
Still, here's my opinion: no, and no.

My thinking is that many use the Shift layer for remapping QWERTY to Dvorak symbols or other layouts, in which case adding Shift is often not desired. Instead of explicitly having to remove Shift in smart macros, the firmware should not add it to begin with.

kareltucek, 2023-07-17:
Well, I think the answer should be no and yes, since if I press shift, I clearly want it to add to my module behavior. (E.g., to select an array of files by two shift clicks in a file manager.)

mondalaci, 2023-07-17:
On second thought, makes sense.

PR #642 (2023-07-17): Fix modifier layer modifiers for non-keystroke actions.
Closes #608.
closed 2023-07-17 by mondalaci