UltimateHackingKeyboard/firmware #609: playMacro defer the last key to next input

yan0lovesha, 2023-05-18:
Repro steps:
1. Create a Command Macro in Agent as below. Bind this Macro to a key. I used Mod+Z
ifShift recordMacro A
ifNotShift playMacro A
2. Open a text editor like notepad.
3. Press Mod+Shift+Z to start record.
4. input "asdf" in sequence.
5. Press Mod+Shift+Z again to stop record.
6. Press Mod+Z to play the recorded macro.

Expected: There will be "asdf" inputted on the text editor.
Actual: Only "asd" were inputted on the text editor.

When I press any key on the keyboard, the last "f" key in the recorded macro then got inputted before the actual key I pressed.
e.g. When I press 'z', there will be 'fz' on the text editor.

kareltucek, 2023-06-19:
Actually, I can't reproduce this.

Are you on the newest firmware? If yes, then please share your UserConfig.

larok00, 2023-07-11:
I am having a similar issue but whole chunks are missing off of the end of my macro. When I start typing again, it will start typing out the remaining parts.

I am on firmware version 9.2.2

Here is an example of how it might happen.
Record macro "hello world" into -> Fn2 + a

Press Fn2 + a:
hello

Press a:
hello worlda

Please note I just gave one example, the behaviour is not deterministic and does not come out the same every time. Sometimes I have to press multiple keys before the keyboard is done typing out the macro and it will start outputting previous chunks of what I wrote, like so:

Press Fn2 + a:
hello

Press a:
hello wo

Press b:
hello worl

Press c:
hello worlda

Press d:
hello worldabc

Press e:
hello worldabcde

If I activate the macro and then hold shift, the macro slowly completes itself while I keep holding down shift. If there is any other experiement you'd like me to run, please let me know. I have one module installed which is the key cluster module. Also I am seeing some weird printing my macro in uppercase sometimes but this is probably an unrelated condition.

kareltucek, 2023-07-12:
Can you please upgrade to the latest firmware, retest, and finally share your UserConfig?

yan0lovesha, 2023-07-12:
I have updated to the latest firmware 10.0.0.0
The behavior is not as same as my initial report. It is still not working. It is more like what @larok00 reported. The behavior is random.

github doesn't support json format. So, I renamed the file to be txt.

[UserConfiguration.txt](https://github.com/UltimateHackingKeyboard/firmware/files/12034004/UserConfiguration.txt)

yan0lovesha, 2023-07-12:
@kareltucek , I can record a video for you if it is still cannot repro on your side.

kareltucek, 2023-07-13:
Still can't reproduce it even with your config. I don't think a video will help.

Can you rule out anything interfering on your side? Like a USB hub, or some software (autohotkey or karabiner elements)?

Do I understand correctly that it happens very often (e.g., in more than say 1 in 10 tries)?

Please test following:
- Create a similar type text "hello world" macro via agent and test whether that macro behaves fine or has the same problem.
- Try any applicable points in https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys . (Yes, I am being paranoid here, but I am always surprised what problems a bad power supply can cause.) Especially:
  - Try if disabling backlight makes any difference. (Is it v1 or v2? What system?)
  - Try with different computer if possible.
  - Try with merged halves.
- Create a macro named `$onInit`, with a command action reading `set keystrokeDelay 10` and then retest.

larok00, 2023-07-17:
Hello, I have realised that for me this issue was specific to Windows 11 Notepad. I thought this would be the simplest app to test in, but for some reason it causes this issue. Adding delays to the macro doesn't seem to help either.

Still, it would be nice to be able to figure out why it's not working in that app but I guess it's less of a problem.

kareltucek, 2023-07-17:
> Still, it would be nice to be able to figure out why it's not working in that app but I guess it's less of a problem.

Well, if you want my opinion on this, then still please try the above points 1 and 3, with 3, increase the delay until you reach 100ms and let me know results. (Is the problem present with keystrokeDelay 100? Is it present for both runtime macro and a regular write text macro?)

kareltucek, 2023-07-30:
I am closing this, since it looks like application-side problem and since no one seems interested much in further debugging... 

As always, feel free to follow up.

closed 2023-07-30 by kareltucek