UltimateHackingKeyboard/firmware #613: Progressive mouse acceleration

Festiu, 2023-05-27:
Hi, I'm trying to make my own buttons to decelerate / accelerate mouse speed, having the same functionality as the default ones via menu - Decelerate / + Accelerate, but with a bypass key with ifAnyMod

```
ifAnyMod final holdKey d // Bypass key
set mouseKeys.move.initialAcceleration 6375
set mouseKeys.move.initialSpeed 500
set mouseKeys.move.baseSpeed 500
delayUntilRelease
set mouseKeys.move.initialSpeed 1800 // Restore speed
set mouseKeys.move.baseSpeed 1800 // Restore speed
```

They work ok, however I feel some kind of a progressive de/acceleration, and I would like to avoid that effect by start moving at base speed immediately, which I can do with the default buttons by the way, but not with my own ones. It's like I was missing some parameter on mouseKeys.move however none of the available ones seems to serve that purpose (initalSpeed, baseSpeed, initialAcceleration, deceleratedSpeed, acceleratedSpeed)

This effect it's tiny, in fact it's nice. However I like moving fast, in a very sensitive way, and I would prefer decelerate instantly in a raw way. Is it possible?

kareltucek, 2023-05-27:
UHK adjusts to target speed by initialAcceleration per second, therefore, your observation is correct. 

Following seems to give instant deceleration:
```
ifAnyMod final holdKey d // Bypass key
set mouseKeys.move.initialAcceleration 1000
set mouseKeys.move.initialSpeed 1
set mouseKeys.move.baseSpeed 1
startMouse decelerate
delayUntilRelease
stopMouse decelerate
set mouseKeys.move.initialSpeed 1800 // Restore speed
set mouseKeys.move.baseSpeed 1800 // Restore speed
```

(It will need some additional tweaking, but I am sure you can manage...)

Festiu, 2023-05-27:
Yes, that's exactly what I needed. Thank you very much

closed 2023-05-27 by Festiu