UltimateHackingKeyboard/firmware #614: Support of register values in setLedTxt

soloidx, 2023-05-29:
Hello, this is not an urgent matter but would be nice:

I am currently creating a basic VIM engine inside the keyboard and I want to display a number from the registry inside the led text panel.

currently `setLedTxt 2000 #22` displays "22" in the led display

kareltucek, 2023-06-19:
Also thinking about `setLedNum 2000 #22`, so that this can be used to output other numeric stuff...

@mondalaci any opinions?

mondalaci, 2023-06-19:
@kareltucek How would `setLedReg` and `setLedNum` differ?

mondalaci, 2023-06-19:
Both commands make sense to me, but is there a way to make commands polymorph instead of introducing per-type command names?

kareltucek, 2023-06-19:
Yes, but the language will need to be refactored to support proper string tokens... ...which admittedly should be done anyways.

But I am not sure if going polymorph is actually a good thing. 🤔

mhantsch, 2023-07-24:
If you refactor the language anyway, perhaps that's a chance to think about the ifGesture/ifShortcut key ids vs. characters again. Number would be key id, and String would be a character to be converted into the matching key id. See also the discussion around https://forum.ultimatehackingkeyboard.com/t/key-id-symbols-wanted/226/35

mondalaci, 2023-07-24:
@mhantsch We've already agreed on key id handling in https://github.com/UltimateHackingKeyboard/firmware/issues/630#issuecomment-1630574561

mhantsch, 2023-07-24:
> @mhantsch We've already agreed on key id handling in [#630 (comment)](https://github.com/UltimateHackingKeyboard/firmware/issues/630#issuecomment-1630574561)

I had missed that one. Thanks.

closed 2023-09-11 by mondalaci