UltimateHackingKeyboard/firmware #615: FTY - Invalid Signature

TeamBrett, 2023-06-01:
My keyboard (v1) is stuck in FTY mode.  I've tried it on 3 different computers with plenty of USB power and I've tried disconnecting power and reconnecting and disconnecting, waiting a deal of time, then reconnecting.  No luck.  When I open the UHK client I get a red message saying "Please power cycle your keyboard (Invalid hardware configuration: Invalid Signature).

[image]

ert78gb, 2023-06-01:
Please check the https://github.com/UltimateHackingKeyboard/agent/issues/2009#issuecomment-1536831370

mondalaci, 2023-06-09:
Closing due to the lack of replies.

closed 2023-06-09 by mondalaci