UltimateHackingKeyboard/firmware #626: Exposing smart macro errors to Agent

mondalaci, 2023-07-05:
Manually dumping the status buffer into an editor is cumbersome. It'd be nice if Agent displayed it.

`UsbCommand_GetKeyboardState()` could expose the error flag, and `UsbCommand_GetVariable` could expose the buffer's content to Agent. Then Agent would display the error in an easy-to-notice top bar with a yellow background.

With the introduction of this feature (both on the firmware and Agent side), the firmware should start throwing errors for deprecated commands, such as `resolveSecondary`, so they can be obsoleted in future major firmware versions instead of keeping around deprecated commands forever.

@kareltucek Thoughts?

PS: We should use consistent naming regarding the buffer. In the reference manual, it's called as both the error and status buffer. 
I think the former is a more apt name.

kareltucek, 2023-07-05:
Thoughts:

- This would be nice if we could compile or validate all macros when config is uploaded to UHK.  We can't at the moment.

- With current state of affairs, I do not see much benefit, therefore assigning a rather low priority to it.

- Ad consistency of naming, indeed.

- Regarding deprecated commands, I don't think firmware should throw deprecation warnings every time such command is used. It would be nice if we could validate all macros contained in the config and throw all at once... ...but that is not possible at the moment.

- This reminds me of an idea that implementing a suitable parser in the agent would allow smart autocomplete, and also the mentioned validation, without cumbersome interaction with UHK itself. (Unlike firmware side, this parser would be generic, pulling relevant grammar as an input from github.) In theory, such generic parser should already exist somewhere in the npm repository...

- There is another issue related to showing errors: at the moment the led display is just overwritten by any text on any led update. As a result, if an error is thrown right before keymap switch, led display ends up showing the new keymap abbrev rather than the ERR... ...we should implement a stateful manager for led display for this reason. 

- I do indeed intend to throw out deprecated commands out at some point, but at this point, they are not a nuisance, so I preffer user-friendliness of keeping them.

mondalaci, 2023-07-05:
An IDE-like Agent parser will help a lot in avoiding errors, but I don't expect it to cover every possible error, and some commands will be obsoleted eventually. I don't consider it a major limitation to only show an error at a time instead of having a robust compiler. Users can resolve errors one by one. I still think my original suggestion has merit.

The current LED display is fundamentally limited, and future UHKs will have more capable displays with a dedicated spot to show parser errors. But if you want to improve the current error display mechanism, one way is to keep switching back and forth between ERR and the actual content of the display in a second or so.

An alternative way of handling deprecations is to have a dedicated deprecation buffer which Agent could display the suggested way or in a more subtle way without triggering ERR.

mhantsch, 2023-07-24:
I like the suggestion a lot. I see it would be used this way: You use the UHK without Agent, and if you encounter the `ERR` display, you just start Agent and it will show you the full error message. Or alternatively, if Agent is already running while `ERR` is encountered, it will immediately pop up the error message.

Much easier than having to have a printerr command configured onto some key, opening an editor, remembering what key the command was, and pasting the error message.

mhantsch, 2023-07-24:
Depreciation could be handled by showing `WRN` (warning) instead of `ERR`.

mondalaci, 2023-07-24:
I like the idea of a dedicated warning buffer and the related WRN mnemonic for deprecation purposes. Agent would display errors and warnings as described with different background colors in the notification bar.

kareltucek, 2023-07-27:
Good one!

PR #661 (2023-08-13): Pull status buffer via usb command
Closes #626.
closed 2023-08-13 by mondalaci