UltimateHackingKeyboard/firmware #630: [Feature proposal] Named variables

kareltucek, 2023-07-10:
I plan to do the following refactor of the language related to tokenization, variables and config values: 

- Introduce implicitly-typed named variables. This would work by having string references pointing into configuration buffer. I.e., 3-byte string reference (2-byte config buffer offset, 1-byte length), another 1 byte for variable type.  These would be allocated when first used and kept in a statically-allocated sored array, accessed via binary-search.
  - assume support for: int, float, bool
  - however, options for future extension are nearly endless...
- Change '#<reg index>' syntax to '$<variable name>'
- Change `setReg <reg number> <value>` to `setVar <variable name>`.
- Introduce numeric expressions as `$(EXPRESSION)`.
- Implement proper string tokenization with bash-like variable interpolation. E.g., `write "String that interpolates $thisVariable.\n"` or `'Produce literal $ character.'`.
- Merge config value and variable namespaces for reading. E.g., allow substituting `write "$module.trackball.baseSpeed"`.
- However, leave separate `set <config value>` and `setVar <variable name>` commands for writing, in order to preserve errors when a typo is made when trying to set a config value.
- Refactor some namespaces:
  - Change `%<index>` syntax to `$queuedKeypress.<index>`
  - Introduce `$keyId.<keyId abbreviation>` and `$scancode.<KEYABBREV>` namespaces. However, allow omitting `$<namespace>.` prefix in commands that expect a specific type. E.g., allow `ifGesture a` instead of `ifGesture $keyId.a` or `tapKey a` instead of `tapKey $scancode.a`
  - Reference number keys in keyId namespace by words ("one", "two", ...) in order to preserve old KEYID syntax. (I strongly believe that unambiguous numeric references have their merits. Also, changing meaning of an existing syntax to something entirelly different while not discarding old syntax validity is just rude.) 

Memory requirements:
- Current 32\*4 byte array for registers will be replaced by a 32\*8 array, i.e., additional 128 bytes of RAM will be consumed.

Your chance to protest is no...

mondalaci, 2023-07-10:
Very impressive and welcome features!

So you're planning variable names of maximum 3 characters, right? I think it's a bit limited, and I'd prefer 8 characters, which would allow for better readability. We could free plenty of RAM by using per-slot MAX_KEY_COUNT_PER_MODULE values.

As for the `#` to `$` syntax change, would you phase out registers and related commands in favor of variables, and would `#` still be used for starting comments?

Please give an example of how `set` and `setVar` would work.

Regarding refactors, I'd love to see overly large (300+ LoC) files getting split up as multiple files.

The only thing I disagree about is the numeric keyId handling. I want to highlight smart macro features in multiple newsletters and blog posts, perhaps on a recurring basis, after this gets implemented. For newcomers, it's natural to use numbers and letters consistently in both the keyId and scancode notations, and the suggested behavior would be very confusing for them. Your list of changes already contains breaking ones, and we could clearly communicate this one, just as all the others, via the newsletter, blog, and forum.

kareltucek, 2023-07-10:
> So you're planning variable names of maximum 3 characters, right?

Nope. 255.
 
> As for the `#` to `$` syntax change, would you phase out registers and related commands in favor of variables

You mean to throw out current registers? Yes.

> and would `#` still be used for starting comments?

Yes.

> Please give an example of how `set` and `setVar` would work.

```
set module.trackball.baseSpeed 0.5 
setVar myCuteVariable 5.5
set module.trackball.speed $myCuteVariable
setVar myCuteVariable $(module.trackball.speed*1.1)
...
```

> Regarding refactors, I'd love to see overly large (300+ LoC) files getting split up as multiple files.

I know... I plan to pursue that when all long living branches are merged.

> The only thing I disagree about is the numeric keyId handling. I want to highlight smart macro features in multiple newsletters and blog posts, perhaps on a recurring basis, after this gets [...]

Well, I disagree on this... If it indeed turns out to be a problem, we can always make those changes.

mondalaci, 2023-07-11:
Let's say a new user wants to reference the Q and 1 key in smart macros.

`tapKeySeq a 1` would work according to his expectations.

`ifGesture a 1` would not work. After banging his head against the wall, he'd eventually figured out that, unlike letters, numbers don't work according to QWERTY.

I feel very strongly that the above is deeply wrong, and keeping backward compatibility is not a good enough reason to keep it this way.

I treat you as a true stakeholder, and I do my best to integrate your contributions as they are, but in this case, I'm willing to diverge from your fork regarding handling numeric key ids. I think it's in everybody's best interest to have consistent behavior across your fork and the official firmware, so I'm asking you to play ball here. I love every other suggestion of yours.

kareltucek, 2023-07-11:
So you also expect special characters to work....

All right, lets have it your way.

mondalaci, 2023-07-11:
Thank you!

Just to avoid any misunderstanding, what do you mean by special characters? I'm unsure. Feel free to provide examples.

kareltucek, 2023-07-11:
Characters like `[]'":;`...

mondalaci, 2023-07-11:
Haven't thought of using these characters directly in smart macro commands for specifying keys/scancodes. Wouldn't they interfere with the existing grammar? Or potentially cause problems when extending the current grammar?

kareltucek, 2023-07-11:
Well, they are allowed in single char scancode notation, so your above argument applies to them.

How much problem they can cause and whether we care is a tricky question. The bottom line is probably that we already parse special characters in scancode abbreviations, so doing the same thing with keyids does not make it much worse.

mondalaci, 2023-07-11:
> The bottom line is probably that we already parse special characters in scancode abbreviations, so doing the same thing with keyids does not make it much worse.

Agreed.

mondalaci, 2023-07-24:
I'd think it mapped to the A key according to QWERTY.

mhantsch, 2023-07-24:
OK, that makes sense. Map everything according to US QWERTY. I got ahead of myself with all my ideas about auto-keymap detection in Agent etc.

closed 2023-09-11 by mondalaci