UltimateHackingKeyboard/firmware #634: oneShot of a layer does not honor modifiers and also not what is defined on a modifier layer (for example the Shift layer)

rpnfan, 2023-07-12:
Example for the first case: oneShot holdLayer fn2 will work as expected when hold, but not when tapped, for example on the m-key (qwerty) I get ] instead of }

attached my config to reproduce the problems described. Using the default keymap I can not access the : character which is defined on the shifted qwerty-c key when trying to use oneShot shift.
[UserConfiguration (2).zip](https://github.com/UltimateHackingKeyboard/firmware/files/12031451/UserConfiguration.2.zip)

kareltucek, 2023-07-13:
> Using the default keymap I can not access the : character which is defined on the shifted qwerty-c key when trying to use oneShot shift.

In this case your macro has to read `oneShot holdKey iS-`. For more information search https://github.com/UltimateHackingKeyboard/firmware/blob/master/doc-dev/reference-manual.md for "composition mode".

> Example for the first case: oneShot holdLayer fn2 will work as expected when hold, but not when tapped, for example on the m-key (qwerty) I get ] instead of }

I see. I can clearly reproduce this, and it puzzles my why this would not work.

kareltucek, 2023-07-13:
Should be fixed by #635.

Thanks for report!

rpnfan, 2023-07-13:
Thanks for the explanation on "composition mode", that works indeed. Also great you found where the other case went wrong and could fix it. :-)

closed 2023-07-14 by mondalaci

rpnfan, 2023-07-15:
Just a question. Does the patch also work, when a dead key is used before? With the US international layout Shift-' will produce the dead " and after pressing for example A (thus again Shift-a) will produce Ä.

kareltucek, 2023-07-16:
I doubt that. I assume the shift would have to be held for next two keys (the dead key and the a key)? And since uhk has no idea which scancode are dead keys according to the host OS, it does not know that in that case one shot the user expects two shots.

rpnfan, 2023-07-27:
Just checked with 10.1 and indeed it works as expected. The shifted dead key is correctly triggered. :-)

PR #635 (2023-07-14): Make oneShot last one more cycle, and prevent oneShot from sleeping.
One more cycle is needed for proper composition of modifiers. (Otherwise end of layer hold may cancel sticky modifiers before their scancode is applied.)

Sleeping is a problem because modified command may enter sleep from which we would not be waken up that one cycle later.

Steps to reproduce:
- have `oneShot holdLayer fn`
- bind shift+c scancode in your fn layer 
- tap your oneshot switcher; tap that shift+c key; observe `c` being produced instead of `C`

Closes #634