UltimateHackingKeyboard/firmware #637: AltGr vs. Alt+Ctrl

rpnfan, 2023-07-13:
On Windows some programs do not recognize Alt+Ctrl instead of the "true" AltGr-key. That means some characters can not be accessed with the UHK, because one can not choose AltGr as a modifier.

Example: with the US international keyboard layout active Outlook does not output ß when a UHK key is assigned to Alt+Ctrl+s.

This is not strictly an error of UHK, but is just the effect of not offering a dedicated AltGr-key.

kareltucek, 2023-07-13:
AFAIK AltGr is just right alt, isn't it?

rpnfan, 2023-07-15:
I double checked and you are right that the right Alt key is _either_ RAlt or AltGr. In that case it is not a firmware problem, but Agent should make clear that RAlt is either that or AltGr for all layouts which make use of it, for example German or US international.

closed 2023-07-15 by rpnfan

mondalaci, 2023-07-15:
RAlt and AltGr are the exact same modifier from a USB standpoint. It's not "either" but the same. I'm writing this to avoid any misunderstanding.

rpnfan, 2023-07-15:
Exactly for that reason it should be made clear that RAlt will not "output" Alt for all layouts which make use of AltGr. Otherwise people will make the same error like I did and try to achieve AltGr function by chosing RAlt _and_ RCtrl in Agent -- which will potentially lead to problems. See example I have given.

mondalaci, 2023-07-15:
What made you originally assume that Alt+Ctrl is supposed to work like AltGr? I don't remember anyone else ever making this assumption, and to my knowledge, Alt+Ctrl never works as a replacement for AltGr.

rpnfan, 2023-07-15:
https://devblogs.microsoft.com/oldnewthing/20040329-00/?p=40003

http://www.kbdedit.com/manual/ex13_replacing_altgr_with_kana.html

https://en.wikipedia.org/wiki/AltGr_key

mondalaci, 2023-07-15:
Thanks for the links. I stand corrected.