UltimateHackingKeyboard/firmware #650: Support advancedStrategy for FN layers

likern, 2023-07-25:
I set up `fn3` as secondary role for  `advancedStrategy` for base layer.
Now I want to do the same for the same key but at different layer.

It will allow me to switch to digits-only layer (`fn2`) and being able to use `+ - * / , . $` at  `fn3`

GUI doesn't allow to switch layer as secondary role for `fn2`, no such option.
I tried to use macros `ifSecondary ...` but it seems also doesn't work.

**Note:** I have master agent and flashed agent firmware
**Note:** I have global `set secondaryRole.defaultStrategy advanced` on `$onInit`

[image]

[image]

[image]

[image]

mhantsch, 2023-07-25:
I believe within Agent, layer switching is only supported from the base layer. You'd have to use macros to achieve what you want (stacking layers).

likern, 2023-07-25:
> I believe within Agent, layer switching is only supported from the base layer. You'd have to use macros to achieve what you want (stacking layers).

I think `switchLayer` just doesn't with advancedStrategy, it least I tried solve this using macroses.

likern, 2023-07-25:
My `$onInit` setup
```
set keystrokeDelay 10

set secondaryRole.defaultStrategy advanced
set secondaryRole.advanced.timeout 350
set secondaryRole.advanced.timeoutAction secondary
set secondaryRole.advanced.safetyMargin 0
set secondaryRole.advanced.triggerByRelease 1
set secondaryRole.advanced.doubletapToPrimary 0

switchKeymap TST
exec configureTrackpoint
```

kareltucek, 2023-07-25:
`ifSecondary` is broken in 10.0.0. Please wait for next release.

If you wish help with any of your macros, please post the macros that you have tried.

closed 2023-07-25 by kareltucek