UltimateHackingKeyboard/firmware #651: Support ctrl+shift using advancedStrategy

likern, 2023-07-25:
According to this comment https://github.com/UltimateHackingKeyboard/firmware/pull/639#issuecomment-1648627706 it should work on master branch with agent firmware.

But it **doesn't work** for me (`tapKey` also didn't work).

- [ ] Just single tap on key `s` doesn't produce key `s`

Overall it doesn't work and fill like `advancedStrategy` setup through UI (which works very well and smooth)

[image]

likern, 2023-07-25:
My `$onInit` setup
```
set keystrokeDelay 10

set secondaryRole.defaultStrategy advanced
set secondaryRole.advanced.timeout 350
set secondaryRole.advanced.timeoutAction secondary
set secondaryRole.advanced.safetyMargin 0
set secondaryRole.advanced.triggerByRelease 1
set secondaryRole.advanced.doubletapToPrimary 0

switchKeymap TST
exec configureTrackpoint
```

kareltucek, 2023-07-25:
You need up-to-date firmware, not agent.

likern, 2023-07-25:
> You need up-to-date firmware, not agent.

I thought if I build agent from source (as in documentation) it will use master firmware. Is it correct?

kareltucek, 2023-07-25:
No.

It downloads the firmware from github according to version that is manually entered in package.json. This version is manually updated and committed to the repository on releases.

You really need to clone the firmware repository and build that.

likern, 2023-07-25:
> No.
> 
> It downloads the firmware from github according to version that is manually entered in package.json. This version is manually updated and committed to the repository on releases.
> 
> You really need to clone the firmware repository and build that.

Get that, will close for now and test on new release

closed 2023-07-25 by likern