UltimateHackingKeyboard/firmware #653: Build custom module re-using existing firmware?

harryhaaren, 2023-07-25:
Hi UHK Folks!

I'm a happy customer, with a few addons to a UHK V1. My only "issue" is that I type `b` with my right-index finger...  and the keyboard splits in half with `b` on the left! I've ordered a few components (steel guide set, spare buttons, etc) to build a `b-button-module` for the right-hand side of the keyboard; progress so far in the image below (mechanically its working 100%).

Currently its attached to an Arduino Leonardo with USB-HID, and a 2nd USB plug connects it to the target laptop, resulting in my `b` working as desired.. :)

But I'd like to improve this to work like the "trackball" (or other right-hand-5-pin-connector-module). I'd like to continue using the offical firmware, and not flash custom firmwares for the future (due to maintenance overhead) . So my current thinking is to "highjack" the trackball module and "pretend" to be one of those over the 5-pin-external connectors.

They talk i2c to the "right-half" of the keyboard, so I think implementing the [i2c slave protocol](https://github.com/UltimateHackingKeyboard/firmware/blob/master/shared/module/slave_protocol_handler.c) is enough to communicate? Specifically, the `SlaveCommand_RequestKeyStates` message would need a proper implementation that reads a GPIO from my custom chip.

Before embarking on coding up a custom i2c implementation of the protocol (for a ESP32-C3-RustDev chip), does the above sound like a sane plan? Is there better ways of achieving integration with a "right-half" that would require less effort? (I'd buy a "b-module" if one existed! :laughing: )

[image]

kareltucek, 2023-07-25:
Sounds sane to me. I think all assumptions are correct, although @mondalaci should probably verify this.

kareltucek, 2023-07-25:
Fortunately, agent support is not needed, as scancode actions can be mapped via macro commands.

mondalaci, 2023-07-25:
I'm closing this issue because it's not an actual task we have to implement/fix, but feel free to follow up.

closed 2023-07-25 by mondalaci

harryhaaren, 2023-07-25:
Thanks for the quick replies!

> Excellent work so far that makes you an honorary Ultimate Hacker. :)

Hah thanks - building on the shoulders of the UHK project & community - but thanks :)

> We'll eventually release a right-side key cluster, but there's no ETA yet.

Looking forward to it - even with a custom one, I'll probably get a "proper" one to look a bit nicer too!

> In the meantime, your best bet is to use a FRDM-KL03Z development board featuring the same MKL03Z32VFK4 processor as the modules, so it'll run the UHK module firmware.

Aha ok -> yes that would "imitate by actually being the same", and then the firmware on the custom `b-module` itself can also be identical to that of the existing modules.

My original plan was to use a different microcontroller (ESP32-C3-Rust) to talk the `i2c` slave-protocol. In doing so, I hoped that the right-hand keyboard firmware could remain the "official" one. Of course, the right-hand firmware needs to somehow know what module is connected - this is where my thinking of "emulating" the `trackball` (or other module) might be a good fit. (My aim is not to reuse the existing firmware for the `b-module` with a NXP FRDM-KL03Z).

I hope to use the ESP32-C3-Rust with the `esp-idf-hal` [Rust crate](https://crates.io/crates/esp-idf-hal) which includes `i2c` support already, including a [i2cSlaveDriver](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/i2c/struct.I2cSlaveDriver.html).

My goal is to build on the `i2c` code above, with the ESP32 as slave device to the "right-hand" half. I gather that the right-hand firmware is essentially polling the modules regularly (e.g. by sending `SlaveCommand_RequestKeyStates` over the i2c link).

With that further info, my "open question" is:
- what is the minimum amount of the SlaveProtocol to implement for getting a keystroke across the `i2c`? My intention is to implement the following:
   - SlaveProperty_Sync (`SlaveSyncString` as per other modules)
   - SlaveProperty_ModuleProtocolVersion (?)
   - SlaveProperty_FirmwareVersion (?)
   - SlaveProperty_ModuleId (?)
   - SlaveProperty_KeyCount (1)
   - SlaveProperty_PointerCount (0)
   - SlaveCommand_RequestKeyStates (returning a single byte with a boolean in LSB, message len 1)

With that amount of protocol implemented, is it expected to communicate successfully with the (official, unchanged right-hand firmware, or will the above require specific changes for "ModuleId"  in RH-firmware? 

> as scancode actions can be mapped via macro commands

You mention mapping here - I have not thought about this aspect just yet - would you expand a bit on how mapping a scan-code-via-macro would look? (I'm fine with hard-coding `b` into the module, as I *know* its the B key I want.. but understand if directly typing "b" from the module isn't possible as the firmware expects to remap!)

kareltucek, 2023-07-25:
> You mention mapping here - I have not thought about this aspect just yet - would you expand a bit on how mapping a scan-code-via-macro would look? (I'm fine with hard-coding b into the module, as I [...]

Modules just report key states. They do not translate them to scancodes, so the right half needs to figure what the actions should be. If you mimic another module (give it same id as an existing module) and have just 1 key, you will be able to simply map it via agent onto the first key of the module (the left module "button", i.e., keyid 192). Otherwise, you would need to map those without direct aid of the Agent GUI. In that case, a macro similar to the following would do the trick, when placed into, e.g., `$onLayerChange any` macro event.

```
set keymapAction.base.15 keystroke left
set keymapAction.base.16 keystroke down
set keymapAction.base.17 keystroke right
set keymapAction.base.8 keystroke up
set keymapAction.base.78 keystroke escape
```

For more information, see https://github.com/UltimateHackingKeyboard/firmware/blob/master/doc-dev/reference-manual.md . (Or ask specific questions.)

But since you seem to go after one key only, this information is probably redundant...

mondalaci, 2023-07-26:
Implementing the listed protocol commands should suffice.

You can actually reuse an existing right-side module id. That way, you don't have to modify the right-side firmware. Agent will show the relevant module in this case.

harryhaaren, 2023-07-28:
Thanks for the prompt replies folks! Got some more work done on the mechanical setup, hoping to get some `i2c` stuff done over the weekend now...

> Modules just report key states. They do not translate them to scancodes, so the right half needs to figure what the actions should be. If you mimic another module (give it same id as an existing module) and have just 1 key <...>

Aha - thanks - yes that all makes sense.

>  You can actually reuse an existing right-side module id. That way, you don't have to modify the right-side firmware. Agent will show the relevant module in this case.

Perfect - this is what I was hoping for! Brilliant, I like this, thanks.

Is the RJ12 cable between the two halves also `i2c`? I'm finding it hard to connect to the "golden push-pin connectors" on the right hand side, and intercepting the RJ12 would be a lot easier... thoughts?

mondalaci, 2023-07-28:
> Is the RJ12 cable between the two halves also `i2c`? I'm finding it hard to connect to the "golden push-pin connectors" on the right hand side, and intercepting the RJ12 would be a lot easier... thoughts?

Yes. Please check out https://github.com/UltimateHackingKeyboard/uhk60v2-electronics

harryhaaren, 2023-07-30:
OK, making some progress here on the hardware and Arduino i2c side of things. Wired up, 4.7kOhm pull-ups in place, so I think HW wise all is wel.

From the SW `i2c` side, but a bit confused as to addressing: there is a list of addresses [defined here](https://github.com/UltimateHackingKeyboard/firmware/blob/fdbef322e0484efe29364afea6f73cbcd90b2a96/shared/i2c_addresses.h#L29) which includes names like `I2C_ADDRESS_RIGHT_MODULE_FIRMWARE` which seems relevant, as that's what I'd like my custom device to emulate, is this correct? (Aka, my Arduino starts its i2c bus with the address `RIGHT_MODULE_FIRMWARE` ?)

I have an Arduino printing `i2c` messages it recieves, it gets a 5 byte message, so I'm working on the assumption that using the "RIGHT_MODULE_FIRMWARE" address for the Arduino i2c-address is correct. The 5-byte message recieved is  `0x02, 0x0, 0x0, 0x0, 0x0`, which when interpreted as a [slave-protocol message](https://github.com/UltimateHackingKeyboard/firmware/blob/master/shared/slave_protocol.h#L21) is the `RequestProperty` with `SlaveProperty_Sync` ID, that seems pretty good? (The CRC is zero -> is that expected/correct?)

My module now wants to reply with the appropriate `i2c_message` response according to the [mesage format](https://github.com/UltimateHackingKeyboard/firmware/blob/master/shared/slave_protocol.h#L64). I have the `i2c_message` built up with `len=3+4`, crc calculated re-using your [CRC impl](https://github.com/UltimateHackingKeyboard/firmware/blob/master/shared/crc16.c) and the data `memcpy()` of the "SYNC" string into the TxMessage.data buffer -> so that's all OK i think...   

Open Question:
What `i2c` address does my module send this "SYNC" message reply to? I can set the address in [Arduino-Wire API](https://www.arduino.cc/reference/en/language/functions/communication/wire/begintransmission/) but I don't know what to set it to...?

mondalaci, 2023-07-30:
The suggested I2C address is correct.

I'm sorry, but I can't allocate time for such custom developments. I'll do my best to reply to simple questions going forward. I suggest using a protocol debugger and comparing your payload with the UHK I2C protocol payload.

harryhaaren, 2023-09-23:
Hi Lazlo,

Totally understand that its not possible to support this type of building/hacking activities, thanks for being up-front and transparent about it.

I was waiting for a logic-analyser for the last weeks (US->Europe shippping, ugh..) but it arrived recently. At the same time, I thought maybe a simpler (better?) idea would be to take the trackball module, and see if the internals of that module allow for putting a `b` button "into" the space near the mounting pins...

So instead of working on low-level firmware, `i2c` debug and such, it becomes a "fun" electronic-wiring & plastic trimming project. Specifically, I "stole" the right-click button on the module to emulate the `b` key (the key-remapping is really useful!) Pictures of the process & end result below here.

Thanks for designing the modules in an easy-to-modify way, and building a great keyboard eco-system! -Harry

__________________

The end result; I'm pretty happy! (The key doesn't sit "perfectly" height or position wise, but its 95% in the right place.. and typing this entire message I haven't missed or accidentally hit a `b` keystroke yet!)

[image]
[image]

The process
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mondalaci, 2023-09-24:
You've done an excellent job; This is one of the greatest mods ever!

Feel free to repost your mod in the UHK forum. I'm sure many forum members would appreciate it, and it'd receive much better visibility than here.

In case you miss the right button, it should be possible to make it work by soldering an enamel wire to one of the free pins of the microcontroller and using it for the new switch while using the right switch as originally intended.

mhantsch, 2023-09-27:
This is really cool. I think all trackball modules should come with the extra key! I had to retrain my fingers when I moved to UHK because I was constantly hitting the empty spot of the `B` key with my right index finger. Eventually my fingers learned, but why not offer the extra key; you have proven that there is room for it!

mondalaci, 2023-09-27:
Good point, although this change may make the individual modules look a bit unsightly.

mhantsch, 2023-09-27:
I don't think the extra key would work for the trackpad -- there is simply no room as you need the touch surface.

The extra key should work for trackpoint and trackball. Maybe offer the most common choice (I assume that's trackball...) with the key. For some users it would be a choice between "do I want the extra key or do I want a trackpoint?".

mondalaci, 2023-09-27:
I also meant the trackball and trackpoint modules, as this design clearly wouldn't work for the touchpad. We'll look into implementing something like this in the future, but we don't want to introduce additional product options.