UltimateHackingKeyboard/firmware #654: First mouse move sometimes omitted when using secondary layer activation with advanced strategy

mhantsch, 2023-07-26:
Highly obscure edge case, but I wanted to document it.

Setup to produce this behaviour:

Firmware 10.1.0
Agent 3.0.0

`$onInit` contains:
```
set secondaryRole.defaultStrategy advanced
set secondaryRole.advanced.timeout 999
set secondaryRole.advanced.timeoutAction secondary
set secondaryRole.advanced.triggerByRelease 1
set secondaryRole.advanced.safetyMargin 50
```

Left space key is configured to type space as primary action, and activate mouse layer on secondary action. No macro on this key.
On the mouse layer `j` is configured to move mouse left.
On the mouse layer `n` is configured to send scancode n.

Now for some tests:

**A)** When I hold left space for more than 1 second until the mouse layer activates, then tap `j`, the mouse pointer moves left for a short distance => good

**B)** When I hold left space, hold 'j', let go of space, let go of 'j', all within 1 second, the UHK sends space followed by j => good

**C)** When I hold left space, tap `j`, let go of space, all within 1 second, nothing happens except that the mouse layer activates briefly => **_move mouse left is missing, the mouse pointer does not move_**

**D)** When I hold left space, tap `j`, tap `j` again, let go of space, all within 1 second, the mouse pointer moves left the same short distance as in case A => **_only one move mouse left is sent but I expected two_**

**E)** When I hold left space for more than 1 second until the mouse layer activates, then tap `n`, the UHK sends n => good

**F)** When I hold left space, hold 'n', release space, release 'n', all within 1 second, the UHK sends space followed by n => good

**G)** When I hold left space, tap `n`, release space, all within 1 second, the UHK sends n => good

**H)** When I hold left space, tap `n`, tap `n` again, release space, all within 1 second, the UHK sends n twice => good

It seems that mouse moves are treated differently than scancode events.

kareltucek, 2023-07-26:
This isn't new behavior, is it? I assume that resolveSecondary/ifSecondary/ifPrimary in versions 9.x and lower behaved just the same.

Also, what exactly is the bug?

Assuming that specification of secondary roles says that press lengths are not preserved, it seems to be the correct behavior. 

The only thing that comes to mind is that you might claim that mouse keys should always move cursor by at least one pixel. However, is movement by exactly one pixel of any use?

mhantsch, 2023-07-26:
I assume the behaviour is the same with resolveSecondary and in previous versions of the firmware.

What I feel is wrong is that the "n" key in case G gets transmitted but the mouse move on case C does not. That is inconsistent. The only difference I see is that one is a mouse HID event and the other is a keyboard HID event. They seem to be treated differently.

kareltucek, 2023-07-26:
The trouble is that distance traveled by mouse key move depends on the time for which it was held. Just tapping a scancode sends that scancode independently on the length of that press.

From the point of view of that mouse key, it was activated for only a very brief moment (1 or 2 ms) after the key is released. That translates to zero or almost zero distance.

mhantsch, 2023-07-26:
I don't understand how the time is different between case A and case C. In both cases I just **tap** the mouse move key. They should both move the mouse by the same amount.

kareltucek, 2023-07-26:
The difference is that in case C the secondary role  (and thus also the action bound to j key) does not get activated until you release the j key.

mhantsch, 2023-07-27:
I see, so I understand the technical reason now. I still don't think it makes sense from user's point of view.

I can envision two ways to change the behaviour:

a) define a minimum movement so a very short tap on a mouse move at least moves the pointer at least a small minimum. At the moment in case C it does not move _at all_.

b) record the timing of the tap on `j`, so when the time machine is able to determine that it is a secondary layer tap, the actual tap timing can be used to determine the amount of mouse move. So basically, record the press and release times, so when secondary layer activation is determined at release time, the amount of mouse move can be calculated from the press time.

kareltucek, 2023-07-27:
From user point of view it still seems to me reasonable that cursor starts smoothly moving when secondary role activates.

B) would mean that if you press and hold the key, cursor does nothing at first and then jumps long distance. Technically it is possible (and, e.g., secondary role resolutions work that way, using original press and release times rather than postponed), but technically complicated and probably not justified.

A) As I wrote above, I can make mouse keys always move by at least 1 pixel, but does one pixel really make a difference for practical use?

mhantsch, 2023-07-27:
Well, there is also option C) do not change anything and accept the current behaviour as intended and "won't fix". :wink: 

As I wrote at the very start of this ticket, it's really an obscure corner case.

For me this is one of those issues that I like to leave open and ponder over time. Sometimes a clever idea comes up after a while.

mhantsch, 2023-07-27:
I think it should move by the same amount that a short tap (with the layer already active) moves it. On my system that is more than 1 pixel (but I may have changed the defaults).

[image]

kareltucek, 2023-07-27:
@mondalaci any opinion from your side?

mondalaci, 2023-07-28:
I find the 1px mouse movement to be impractical, and it's highly affected by operating system settings.

Maybe, for practical purposes, dual role keys should immediately transition to the secondary role regardless of the timeout for mouse actions.

mhantsch, 2023-07-28:
1px is definitely impractical.

@mondalaci Your suggestion for immediate activation of secondary role for keys with mouse action won't work, for example if I use secondary action on space key to activate mouse layer and ijkl as mouse move keys. That will conflict with normal typing as space followed by any of those keys could often happen at the start of words. You really need the advancedStrategy resolution to distinguish between primary and secondary here.

Since it's based on a "time machine" I believe for a proper solution the timing would need to be considered. Once the secondary role triggers, it would need to go back and behave "as if" it had been activated when the second key was pressed down. I realise that's hard to implement just for this edge case.

mhantsch, 2023-07-28:
@kareltucek You'd need something like an integrated acceleration formula that calculates for a given "past time" how much the acceleration would have moved the mouse so far, and send a mouse move for that distance, plus set the current speed so that it can continue to accelerate correctly from that point (unless the key has already been released, of course, in which case it's just the initial "past time" move).

kareltucek, 2023-09-01:
Here is POC of the suggested functionality [uhk-firmware-10.1.0-timemachined_mouse_keys.tar.gz](https://github.com/UltimateHackingKeyboard/firmware/files/12491959/uhk-firmware-10.1.0-timemachined_mouse_keys.tar.gz) . 

Please, @mhantsch, @mondalaci, and whoever else interested in this issue, try it and let me know what you think. 

Personally, I am not convinced that the behavior is intuitive and desirable. Feels more like an unintended glitch. I am not convinced we actually want it.

Testing instructions in case they are needed:

- bind the following macro on your x key:
    ```
    ifSecondary advancedStrategy final holdLayer mouse
    tapKey x
    ```
- use your mouse keys using this new x secondary layer switch, by simultaneously hitting (and optionally holding) x with some of your mouse keys.
- observe the cursor jump in the direction after a short delay.

mondalaci, 2023-09-08:
@mhantsch Please share your opinion on Karel's POC. I'd like to have a consensus on this and resolve/close this issue.

I'm sorry, but this is such an obscure and relatively complicated edge case that I'd rather devote time to higher-priority issues instead of testing it.

mhantsch, 2023-09-09:
@mondalaci I'll test it this weekend. I was away on a business trip (without my UHK).

mhantsch, 2023-09-09:
@kareltucek @mondalaci OK here we go...

Definitely works differently now, but not exactly to my liking. There's a big jump now and it feels the jump is larger than what I would expect. Are you using the start timing when the secondary role layer switch key (in my case: left space) was pressed, or when the actual mouse movement key (in my case: j) was pressed down? It feels like the initial move is too large and not in line with the timing of the second key press. Hope my explanation makes any sense.

To compare:

Case 1) hold down space until mouse layer activates, then briefly press and release j. Mouse moves a tiny bit, maybe 5-10 pixels.

Case 2) press down space and immediately follow by briefly press and release j. Mouse moves a large bit, maybe 30-40 pixels.

IMHO, the mouse should move the same distance, because the timing of the j key was the same. It should be independent of the space key -- that one only influences whether and when the secondary layer gets triggered.

The current behaviour of this POC is not smooth. Right now, I would prefer the original behaviour of the standard firmware.

I agree this is a minor flaw, and you probably do not want to spend a lot of time on this over some of the other topics in your backlog. Prioritise as you like. I'm also happy to give you feedback on other POCs if you throw them at me.

kareltucek, 2023-09-09:
When the actual mouse movement key was pressed down.

The trouble may be the release rather than the press, as real time is used as release time, i.e., the time when the secondary role activates.

mondalaci, 2023-09-09:
I think it's not worth spending more than about 2 hours on this niche use case, and fixing some other minor issues, such as #655 or #673, would give more value to customers.

kareltucek, 2023-09-09:
Mmmm, the mouse key processor does not have information about keys that activated it available, which means that I cannot easily retrieve the actual release time... 

This means that pursuing this further is probably indeed not worth the trouble, as I suspect that the user experience would not improve much since the delay and jagged jumping of the cursor are inherent to the problem.

closed 2023-09-09 by kareltucek