UltimateHackingKeyboard/firmware #655: cannot set rgb backlight on key cluster module with UHK v1

mhantsch, 2023-07-28:
An UHK60v1 with key cluster module.
Agent 3.0.1, firmware 10.1.0.

Backlighting mode configured as 'functional backlighting' in Agent.

I am trying to change the key cluster backlight depending on the layer.

Relevant part of `$onInit:`

```
set backlight.strategy constantRgb
set backlight.constantRgb.rgb 64 64 64
set leds.fadeTimeout 300
```

Default keymap 'CMX'.

Relevant part of `$onKeymapChange CMX`

```
set backlight.keyRgb.base.128 128 128 128
set backlight.keyRgb.base.129 128 128 128
set backlight.keyRgb.base.130 128 128 128
set backlight.keyRgb.mod.128 0 128 0
set backlight.keyRgb.mod.129 0 128 0
set backlight.keyRgb.mod.130 0 128 0
set backlight.keyRgb.mouse.128 0 128 128
set backlight.keyRgb.mouse.129 0 128 128
set backlight.keyRgb.mouse.130 0 128 128
set backlight.keyRgb.fn.128 128 0 0
set backlight.keyRgb.fn.129 128 0 0
set backlight.keyRgb.fn.130 128 0 0
set backlight.keyRgb.fn5.128 128 128 0
set backlight.keyRgb.fn5.129 128 128 0
set backlight.keyRgb.fn5.130 128 128 0
```

I am expecting to see the colour of the key cluster backlight to change when I activate different layers. However, nothing happens. The key cluster just keeps lighting in the same colour all the time.

I also tried to configure per-key backlighting in Agent, and configure different backlight colours for the key cluster keys on each layer. However that also did not achieve anything.

Apparently, rgb backlight control does not work with UHK60 v1, not even on the key cluster (which supports full rgb backlight).

matqua, 2023-08-18:
I too have the same issue.

```
Agent version: 3.0.2
Right keyboard half firmware: UltimateHackingKeyboard/firmware:10.1.0
Left keyboard half firmware: UltimateHackingKeyboard/firmware:10.1.0
Key cluster firmware: UltimateHackingKeyboard/firmware:10.1.0
Trackball firmware: UltimateHackingKeyboard/firmware:10.1.0
```

chrysn, 2023-12-09:
The issue is still present with Agent 3.2.1 and firmware 10.4.0.

mondalaci, 2023-12-09:
@kareltucek Would you look into this? It seems like a relatively popular issue, so it'd be worth resolving it.

PR #739 (2024-01-05): Implement backlight for v1.
Closes #569.
Closes #655.
closed 2024-01-05 by mondalaci