UltimateHackingKeyboard/firmware #656: nrecognized command: untoggleLayer

likern, 2023-07-29:
Before submitting a new issue, make sure to do the following:

1. Install the latest Agent:
   https://github.com/UltimateHackingKeyboard/agent/releases/latest
2. Use Agent to update to the latest firmware:
   https://github.com/UltimateHackingKeyboard/firmware/releases/latest
3. Our contact form will suggest you solutions, so copy-paste your message at https://ultimatehackingkeyboard.com/contact-us - Please don't post customer support issues here.

kareltucek, 2023-07-29:
Should be `unToggleLayer`. (Yes, macro commands are case sensitive.)

closed 2023-07-29 by kareltucek

likern, 2023-07-29:
Sorry, my bad. I've verified before creating new issue, but it turns out that didn't turned on match case while searching.