UltimateHackingKeyboard/firmware #658: Serious problem, UHK connected - random behavior (mouse?)

rpnfan, 2023-07-30:
[UserConfiguration (2).zip](https://github.com/UltimateHackingKeyboard/firmware/files/12210030/UserConfiguration.2.zip)

Using my default config I experience very strange behavior. For example while typing the menu shortcut is fired. First I thought that I was accidently triggering that, but now I am sure it is a problem of the UHK. Another problem I see is that when I click on a link in the browser (both in Opera and also in Firefox) the link will be opened in a new tab, which is wrong and not according the HTML code. This happens randomly and I can not force it to happen. But when it starts to happen it is always the case and can be reproduced. A quick but momentary fix is to reset the UHK by unplugging and reconnecting it. I have verified that now a few times. I never had such problems with a standard USB keyboard and am sure the UHK is messing up something. Maybe typing fast (close to 100 wpm) can cause this hick-up? I am on a new layout and mostly still slower.

kareltucek, 2023-07-30:
Interesting...

When this happens next time, please:
- Use some keyboard visualiser ( https://keyboardchecker.com/ or https://config.qmk.fm/#/test ) to see what exactly UHK does and if you can spot something that could help pinpoint the issue.
- Run following macro and post output. I guess it won't tell us anything interesting, but I don't have any better idea...

```
set keystrokeDelay 30
diagnose
printStatus
```

Can you rule out interference of that fn2 layer? (Since it has no corresponding indicator led, its activations may easily go unnoticed...)

kareltucek, 2023-08-01:
Regarding what to test, please, get some inspiration at https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys . 

Especially power issues are always a suspect when it comes to weird problems, so disabling backlight is definitely a thing to try...

Regarding alt-tab, could the switch be faulty? With the hot-swap switches, it (for me) usually helps to pull the switch, bend both legs a little and put it back.

(Also, I will be away for a few days, so sorry for not responding in advance...)

rpnfan, 2023-08-01:
Thanks for the ideas. I do not think that the switches are faulty. I get the same Alt-Tab problem with the original keys and as well with the binding to two other keys.
Power is also not the problem. I now disabled the lighting (which I have pretty low anyways) and connected to a hub with 1.5 A output.

I said I ruled out the modules, but in fact I meant the trackpoint one. Just testing without the left module. Maybe that was the cause!?

kareltucek, 2023-08-01:
> I said I ruled out the modules, but in fact I meant the trackpoint one. Just testing without the left module. Maybe that was the cause!?

That seems unlikely. (Of course, testing it to be sure is welcome.)

> Also important to consider that I did not have those strange problems when I started to use the UHK some weeks ago.

Do you think the problem is specific to a specific firmware version? 

If so (i.e., if you test it and you are sure of it), we can try bisecting towards a specific commit...

rpnfan, 2023-08-01:
Good point that a specific firmware might have broken whatever is causing the issue. Then I will go backwards till it seems solved. At the moment with both HW modules detached I have not had problems so far. We will see if it stays like that.

mondalaci, 2023-08-01:
If some of the modules are the culprit, try https://ultimatehackingkeyboard.com/repair/fastening-module-steel-guide-screws

rpnfan, 2023-08-02:
Just had the problem in the web browser again, without any modules connected. So they can also be ruled out. Next step is downgrading the firmware I guess. :-(

rpnfan, 2023-08-05:
Now back to 10.1 I just had the opens-in-new-tab-problem again. I also noted that I could not scroll with the mouse wheel any longer, interestingly that was the case with my external mouse _and_ with the left module. I played around a bit and the problems were reproducible about maybe half a minute or a minute. Before I could revert to another firmware the problem disappeared and both opening a link in the same browser and also the scrolling function were working again. Just wondering which step might now be most helpful to isolate the problem cause...!?

mondalaci, 2023-08-06:
Please report on your i2cRecovery counter according to https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys

rpnfan, 2023-08-06:
The output now was
2023-08-06 20:56:01 I2cRecovery: 2

I could neither force the value to go up (eg by using the phone very close to the keyboard) nor was I seeing a value of 0. I restarted Agent a few times and repeated the test.

Then I switched to another (powered) hub and repeated the test and finally got 0. Then I repeated the test a few times, both connected to the powered hub and also to the unpowered hub (where the UHK is connected by default). In all tests I only got 0, even when using the non-powered hub where before I got the 2 value.

I will now stay on 10.1 and leave the UHK connected to the powered hub and will see what happens. Or should I test something else?

mondalaci, 2023-08-06:
There's nothing else on my mind to be tested, and I don't know the cause of the issue yet.

Please keep us updated.

rpnfan, 2023-08-07:
I redid the test and now got an error of 1. That was connected to a powered hub. Redoing the test (after restarting Agent) I get the same error. When I click on the copy icon the content is _not_ copied to the clipboard.

[image]

kareltucek, 2023-08-07:
@mondalaci what is meaning of the `/debug-i2c.ts` script output? 

```
2023-08-07 13:55:27 I2cRecovery: 2
```

Is `get-i2c-health.ts` output interesting / relevant to the problem?

rpnfan, 2023-08-07:
2023-08-07 16:36:58 I2cRecovery: 3
with no change I am aware of, still connected to the powered hub

mondalaci, 2023-08-07:
@kareltucek `i2cRecovery` is the sum of the timeouts of `get-i2c-health.ts`, which is sufficient for user debugging.

@rpnfan I'm at a loss of what's happening behind the scenes, as the mentioned `i2cRecovery` values are low and shouldn't cause issues.

How often do you experience these issues? If you use any unusual configuration options, such as the advanced resolution strategy, please disable them, keep everything very basic, and try to reproduce the issues.

kareltucek, 2023-08-07:
> Now I am back to 10.1 (for 1 day) and I just had the opens-in-new-tab-problem again. I also noted that I could not scroll with the mouse wheel any longer, interestingly that was the case *with my external mouse* and with the left module.

Affecting external mouse is very suspicious. It seems like UHK must have produced some modifier for that half a minute that changed the meaning of that scrolling. (Ctrl+left click causes opening in new tab in firefox. A modifier could also explain that scrolling stops working, although I would expect ctrl+scroll to produce zoom in/out.)

It would be nice if you could confirm or refute this hypothesis. If it is correct, I can try to modify the firmware so that we can find which key produces such a modifier.

rpnfan, 2023-08-07:
Ah, that was a good pointer. We are on to something here. I just tested ctrl-leftClick and you are right that this is the shortcut to open a link in a new tab. Then I realized my report was wrong, a link is opened in a new window and not a tab! And as luck wants it this effect just happened again. The shortcut for that is Shift-leftClick! And that also explains why I sometimes had the problem that I could not click on text without marking it, because Shift was active as it seems. And indeed I now had all three problems: 1. link opens in new window 2. scrolling does not work and 3. clicking on text marks it.

You remember that another problem was sometimes the context menu popping up. This can be triggered by the menu key _or_ Shift-F10!!!  Maybe also other ways?

Firmware 10 introduced one-shot-keys and I think here something can go wrong sometimes. 

I can now go back to 9.3 and see if I really do not experience the problems there (for even more than 2 days) or stay at 10.1 and test something else. I think I already tried disabling one-shot-shift, but am not 100% sure about that.

Attached my current config. I now added a standard shift key (the one on the right side) to be able to play around with it, when the effect happens again. As you see in my key map I do not use the original right shift key for shift normally.

[UserConfiguration.zip](https://github.com/UltimateHackingKeyboard/firmware/files/12285333/UserConfiguration.zip)

kareltucek, 2023-08-07:
> What are those strategies? I am not sure if I use those.

Secondary roles, `resolveSecondary` etc.

> UserConfiguration.zip

 `resolveSecondary` and `oneShot` are the only "unusual" features I see in the config. You are definitely not the only user of `resolveSecondary`, but you are quite likely to be the only one actually using `oneShot`. 

You can do something like the following to see when oneShot is operating:

```
set backlight.constantRgb.rgb 255 0 0
oneShot holdKey iS-
set backlight.strategy functional
```

rpnfan, 2023-08-08:
Thanks for the tip. This is helpful. I have changed the macro accordingly. With my key (ID 27) I can unshift a one-shot shift, which might be handy for testing.

rpnfan, 2023-08-09:
I have started a log book. First entry:

2023-08-08
23:00 context menu popup 1x in ngram/ Opera

This morning I checked the error count and is
2023-08-09 08:37:28 I2cRecovery: 40
After replugging it goes back to 0.

mondalaci, 2023-08-09:
Please use a configuration that is as close to the stock configuration as possible and doesn't contain smart macros, not just one-shot keys, assuming that fancy smart macro features are responsible for this issue.

rpnfan, 2023-08-09:
Do you mean for test purpose I should disable the 3 macros I use on the base layer?

1. one-shot-shift
2. one-shot-F2
3. combined space-mod key (code see below)
4. (and the mouse layer switch on the cluster module)

```
# tap: space, hold: mod (Max approach)
resolveSecondary 250 primaryaction secondaryaction 
primaryaction: final tapKey space
secondaryaction: final holdLayer mod
```

rpnfan, 2023-08-09:
I disabled all smart macros in all layers, will see what happens or not.

rpnfan, 2023-08-09:
I was using Alt-Tab (with the standard keys, not via the mod-layer) and while holding Alt and pressing Tab the keyboard seemed to skip. At the same time I saw in the log errors increasing. Now, while I type this message I get more errors again (10 and 11)

2023-08-09 10:18:25 I2cRecovery: 4
2023-08-09 12:25:03 I2cRecovery: 5
2023-08-09 12:26:21 I2cRecovery: 6
2023-08-09 12:26:33 I2cRecovery: 7
2023-08-09 12:26:49 I2cRecovery: 8
2023-08-09 12:27:18 I2cRecovery: 9
2023-08-09 12:32:05 I2cRecovery: 10
2023-08-09 12:32:35 I2cRecovery: 11

Might be a different problem, but possibly this is the cause that randomly I might get errors (and now the next one, 12...)

mondalaci, 2023-08-10:
Are you using the stock bridge cable?

You can try the `set i2cBaudRate 50000` command, which halves the baud rate and should make the internal communication of your UHK twice as robust as in its default state at the expense of reduced responsiveness.

rpnfan, 2023-08-10:
I am using the stock cable and followed the advices on the relevant knowledge base. I had my mobile phone on the table, but now have it 2 meters away also (which btw I do not find a good permanent solution). Still the problems do not disappear.

I just tried `set i2cBaudRate 50000`, which does not make a difference. While I am typing this the errors raised from 0 to 6 already -- all with the new setting in the onInit macro.

mondalaci, 2023-08-10:
Whenever the i2cRecovery count increases, some interference happens. One or some of the points of https://ultimatehackingkeyboard.com/repair/fixing-misbehaving-keys should be the cause.

rpnfan, 2023-08-10:
Devices I have close to the keyboard are:
1. Mini PC (65 cm)
2. External soundcard (50 cm)
3. Amp (150 cm)

Like said I followed the tips already. I do not know what I should do now. :-( 

The only thing which I did not try is the foil, swapping switches and connecting directly to the PC. I now connect directly for testing, but for practical use I need to be able to connect to a USB-hub, because my PC does not have many ports and most important I use the keyboard with two computers, where the built in USB ports of my monitor serve as sort of a KVM switch. That has never made any problems with a "dumb" keyboard.

My current thinking is that I have two unrelated problems. One the interference and second with smart macros.

rpnfan, 2023-08-14:
The next update. I ruled out smart macros, phone being to close, USB port/ hub. I could switch back to firmware 9.3 now and monitor if I experience still i2c errors. That seems to be the only or main problem. A bit frustrating I must say...

The last part of my error log is below - for reference

```
10.1 without smart macros, except base-space-mod
09.08.2023 10:00:38
changed to this config

 09.08.2023 10:08:46
pop-up menu in browser text field (github)

10.1 without any smart macros
09.08.2023 10:09:29
changed to this config                             

10.08.2023 11:16:58
interference errors, not solved by set i2cBaudRate 50000
→ now connected directly to the PC

14.08.2023 14:24:47
pop up menu in browser (ngram), 
i2c errors increasing, phone and mobile phone close (15 cm and then 2 cm)
2023-08-14 11:33:31 I2cRecovery: 5
2023-08-14 14:19:24 I2cRecovery: 6
2023-08-14 14:19:37 I2cRecovery: 7
2023-08-14 14:19:56 I2cRecovery: 8
2023-08-14 14:20:09 I2cRecovery: 9
2023-08-14 14:20:13 I2cRecovery: 10
2023-08-14 14:20:24 I2cRecovery: 11
2023-08-14 14:21:57 I2cRecovery: 12
2023-08-14 14:25:04 I2cRecovery: 13
2023-08-14 14:25:24 I2cRecovery: 14
2023-08-14 14:26:29 I2cRecovery: 15

14.08.2023 14:27:18
→ phone put 1 m away, but still increasing errors
2023-08-14 14:28:37 I2cRecovery: 16
2023-08-14 14:28:38 I2cRecovery: 17

14.08.2023 14:30:14
→ phone put 2 m away, still increasing errors ... 
2023-08-14 14:31:22 I2cRecovery: 19
and again when typing (after a pause)
2023-08-14 15:21:36 I2cRecovery: 20
2023-08-14 15:22:10 I2cRecovery: 21
...
errors occur only while typing and not, when the PC idles
2023-08-14 15:35:39 I2cRecovery: 31
2023-08-14 15:38:03 I2cRecovery: 32
2023-08-14 15:39:23 I2cRecovery: 33
2023-08-14 15:40:22 I2cRecovery: 34
2023-08-14 15:40:25 I2cRecovery: 35
2023-08-14 15:40:26 I2cRecovery: 36
2023-08-14 15:40:57 I2cRecovery: 37
2023-08-14 15:41:22 I2cRecovery: 38
2023-08-14 15:42:30 I2cRecovery: 39
2023-08-14 15:44:07 I2cRecovery: 40
2023-08-14 15:44:15 I2cRecovery: 41
```

mondalaci, 2023-08-14:
@rpnfan Please contact support at ultimatehackingkeyboard dot com, and we'll replace your UHK.

rpnfan, 2023-08-14:
Also with 9.3 I get errors and strange effects like already described. I will contact the support.

2023-08-14 16:07:45 I2cRecovery: 0
2023-08-14 17:06:21 I2cRecovery: 1
2023-08-14 17:10:06 I2cRecovery: 2
2023-08-14 17:44:15 I2cRecovery: 3
2023-08-14 18:05:23 I2cRecovery: 4
2023-08-14 18:12:39 I2cRecovery: 5
2023-08-14 21:09:47 I2cRecovery: 6
2023-08-14 21:31:27 I2cRecovery: 7
2023-08-14 21:32:34 I2cRecovery: 8
2023-08-14 21:35:20 I2cRecovery: 9
2023-08-14 21:38:23 I2cRecovery: 10
2023-08-14 21:57:27 I2cRecovery: 11
2023-08-14 23:22:04 I2cRecovery: 12
2023-08-14 23:22:10 I2cRecovery: 13
2023-08-14 23:22:22 I2cRecovery: 14
2023-08-14 23:22:28 I2cRecovery: 15
2023-08-14 23:23:08 I2cRecovery: 16
2023-08-14 23:23:09 I2cRecovery: 17
2023-08-14 23:23:10 I2cRecovery: 18
2023-08-14 23:23:12 I2cRecovery: 21
2023-08-14 23:24:51 I2cRecovery: 22
2023-08-14 23:24:54 I2cRecovery: 23
2023-08-14 23:24:59 I2cRecovery: 24
2023-08-14 23:25:01 I2cRecovery: 25
2023-08-14 23:25:33 I2cRecovery: 26
2023-08-14 23:25:37 I2cRecovery: 27
2023-08-14 23:25:52 I2cRecovery: 28
2023-08-14 23:25:55 I2cRecovery: 30
2023-08-14 23:26:03 I2cRecovery: 31
2023-08-14 23:26:04 I2cRecovery: 32
2023-08-14 23:26:07 I2cRecovery: 33
2023-08-14 23:26:12 I2cRecovery: 34
2023-08-14 23:26:16 I2cRecovery: 35
2023-08-14 23:26:17 I2cRecovery: 36
2023-08-14 23:26:19 I2cRecovery: 37

mondalaci, 2023-09-11:
This should be a hardware or EMI-related issue, so I'm closing it.

closed 2023-09-11 by mondalaci