UltimateHackingKeyboard/firmware #668: [Feature proposal] Macro validation

kareltucek, 2023-08-17:
Assume I write a solution that can validate smart macros, same as the config is validated via a dry run. 

It would mean some nonnegligible amount of code clutter.

Do you find it worth the effort and the added maintenance burden (due to the code clutter)?

kareltucek, 2023-08-17:
Pros:
- The user gets warnings about errors and deprecations immediately upon saving the configuration.
- Same deprecation warning is not repeated with every use of the command. (I assume that warnings would be enabled only during the dry run.)

Cons:
- Code clutter and related decrease in code readability.
- Obviously costs time to implement.

mondalaci, 2023-08-17:
Depends on the amount of code clutter. In my view, about a hundred extra lines may be worth it, but not a thousand lines.

Increased power consumption is also a consideration. Would you run this validation on firmware bootup and configuration saves?

kareltucek, 2023-08-17:
Like a one additional return somewhere in implementation of every command (~100 lines in total?), plus a ~50 line function containing take care of traversing all the commands and dry-run them.

I assume I would run this validation on configuration saves, and maybe on bootup. It seems to me that config validation does not bring much added value on bootup and so if power consumption is a concern.

mondalaci, 2023-08-17:
Can you prefix the warnings/errors with the related macro names? If so, how much would prefixing increase clutter, assuming it'd increase it at all?

mondalaci, 2023-08-17:
How `actionIndex` and `commandAddress` differ from line numbers?

kareltucek, 2023-08-17:
[image]

mondalaci, 2023-08-17:
Gotcha, thanks. Line numbers would be much more useful than command addresses. How much extra clutter to expose them?

kareltucek, 2023-08-17:
Either
- 30 lines of ugly and possibly buggy code that will be pain to maintain
- or 30 bytes of memory and 10 lines of code which will be lesser pain to maintain

mondalaci, 2023-08-17:
Picking the second option is a no-brainer here. Please make it happen.

kareltucek, 2023-08-17:
Allright!

(I will simply assume it is a yes also to the original proposal of this ticket 😃.)

mondalaci, 2023-08-17:
It is. :)

When there are smart macro related errors/warnings, configuration saves will always succeed, and then the errors and warnings will be put into the status buffer, right?

mhantsch, 2023-08-18:
Would you make Agent automatically retrieve the status after config save, so user gets an immediate response within Agent whether their macros are good? (Instead of having to look at the UHK display, then do some printStatus magic...)

kareltucek, 2023-08-18:
Yes!

mondalaci, 2023-08-18:
@mhantsch This will naturally happen as part of the planned Agent feature that exposes errors and warnings to the user.

PR #671 (2023-08-24): Validate macros when new config is flashed.
Closes #668.

Alright, it is a bit more than 100 lines 😅, but most of it is just simple returns. No devilish algorithms.
closed 2023-08-24 by mondalaci