UltimateHackingKeyboard/firmware #672: Misbehavior of a macro with conditionnals

nikophil, 2023-08-22:
Hello,

(first, sorry if I'm not posting at the right place, if so, please, show me where I may post this)

hi have some macros that I use to display french characters:
Here is an example of a macro:
```
set macroEngine.extendedCommands 1
ifShift suppressMods tapKeySeq CS-u e 8 space  # displays "è"
ifCtrl suppressMods tapKeySeq CS-u e a space  # displays "ê"
ifNotAnyMod tapKeySeq CS-u e 9 space  # displays "é"
```
I have assigned this macro to the key `e` in the layer `fn 2`. So I expect a `è` to be displayed when I press `fn 2` + `shift` + `e`, but I also expect the same char to be displayed when pressing `shift` + `fn 2` + `e` in this order. But it displays a `é` (the one with `ifNotAnyMod`). This is a little bit inconvenient, since I need to pay attention to the order of the keys I press.

I'm pretty sure this problems comes from the last update I made. I'm currently using firmware 10.1.0 and agent 3.0.2 - sorry I have no clue of what was the previous versions.

Not sure if this is a real issue, or the normal behavior... Is there any way to modify the macro in a way that the order of the keys is not important?

thanks!

kareltucek, 2023-08-22:
The macro is correct. There must be another problem in your config. Is your shift key bound to shift in both layers?

Can you export and post your UserConfig?

nikophil, 2023-08-22:
Hi, thanks for your answer. I actually was wrong in my explanation, it is in the other way:
- `shift` + `fn 2` + `e` => displays the right char
- `fn 2` + `shift` + `e` => displays the right char

> Is your shift key bound to shift in both layers?
ha thanks! it was actually NOT bound in the new layer. this was actually not coming from the upgrade 😅

thank you!

closed 2023-08-22 by nikophil

kareltucek, 2023-08-22:
Happy to help!