UltimateHackingKeyboard/firmware #673: Momentary random colors upon booting

mondalaci, 2023-08-23:
When powering up the UHK, random key colors can be seen for a moment. This is slightly disturbing and should be avoided.

kareltucek, 2023-09-17:
Looks like factory mode is partially broken since following commit:

```
commit 481698e35f4196620a6884f764cbeaddf61a7e4d
Author: László Monda <laci@monda.hu>
Date:   Wed Feb 22 10:45:21 2023 +0100

    Put CurrentKeymap into the .data2 section because it wouldn't fit into .data1 with the RGB values included.
```

Scenario 1:
- Make config Validation fail (e.g., by flashing incompatible UserConfig), or manually set `IsFactoryResetModeEnabled = true;`
- Powercycle UHK.
- Observe that factory mode does not work. (Segment display reads FTY, backlight shows random colors, key taps produce no characters on screen)
- Press the factory reset button.
- Observe that behavior has not changed.

Scenario 2:
- Let your UHK load valid UserConfig.
- Press the factory reset button.
- Observe that factory mode works fine.

Scenario 3:
- manually set  `IsFactoryResetModeEnabled = true;`
- run `make flash`. (UHK restarts without powercycling and goes into factory mode.)
- Observe that factory mode works fine.
- Press the factory reset button.
- Observe that factory mode still works fine.
- Powercycle UHK.
- Observe that factory mode does not work.

----

@mondalaci does this ring any bells?

Is there anything special about how the data2 segment is initialized?

mondalaci, 2023-09-17:
This doesn't ring a bell to me.

If there's anything special about the data2 segment, the https://community.nxp.com/ forum may be able to help.

Any further questions or findings, please do let me know.

kareltucek, 2023-09-17:
More observations:

- I have been wrong when claiming that FTY works fine in some cases. In practice, last loaded keymap seems to remain loaded in memory. So it seems that the data2 segment never gets initialized, with one exception:
- Data2 segment seems to be initialized fine during the first boot after flashing. (Why should this be? What does this signify?)

mondalaci, 2023-09-17:
Chances are data(2) segment(s) are uninitialized by default for performance reasons and they should be explicitly initialized when required.

kareltucek, 2024-01-10:
No luck with a nxp inquiry... ( https://community.nxp.com/t5/Kinetis-Microcontrollers/m-data-2-not-initialized/m-p/1785711 )

I guess it is time for the suggested workaround of initializing it explicitly ourselves?

mondalaci, 2024-01-10:
Let's ask our NXP contact before using the workaround. I'll provide his contact via email very soon.

It's worth mentioning that the part number of the MCU is MK22FN512VLH12. The K22P121M120SF7 is some sort of identifier in the datasheet that covers multiple MCUs of the same kind.

kareltucek, 2024-01-10:
I think I have a working version, thanks to bobpaddock's today reply. It edits lib/KSDK_2.0_MK22FN512xxx12/devices/MK22F51212/gcc/startup_MK22F51212.S. Should I post a PR into the UHK submodule repository, or should I create a copy of that file directly in firmware repository?

mondalaci, 2024-01-10:
The former should be the correct way if the fix belongs to a file in the KSDK repo.

PR #742 (2024-01-11): Initialize .m_data_2 RAM segment.
Closes #673 

Steps to reproduce:
- Press the button on the back side to jump into factory keymap.
- Observe that the keymap does not work, including random lighting, as it is initialized randomly.
closed 2024-01-11 by mondalaci