UltimateHackingKeyboard/firmware #677: Add moduleId argument to UsbCommand_GetDeviceProperty() : DevicePropertyId_FirmwareChecksum

mondalaci, 2023-09-04:
Please add a `moduleId` byte argument to `UsbCommand_GetDeviceProperty()` : `DevicePropertyId_FirmwareChecksum`, so that it returns the MD5 of the right half for `0` and of the left half and modules for other values.

The above is needed for Agent to compare these values with the ones provided by `UsbCommand_GetModuleProperty()` : `ModulePropertyId_FirmwareChecksum`.

kareltucek, 2023-09-04:
Not sure if I understand...

So you basically want right side firmware to know (and provide) checksums of the other module firmwares of the same version, is that right?

Why exactly do you want to compare these checksums instead of just comparing whether the packed checksum matches the flashed checksum or not?

mondalaci, 2023-09-04:
This way, Agent can solely rely on the information provided by the firmware to display whether the slave firmwares are up-to-date, even if their version differs but their md5 matches.

kareltucek, 2023-09-04:
I see, makes sense.

PR #679 (2023-09-06): Extend GetDeviceProperty to return other module checksums.
Closes #677.

Not tested. Consider testing blocked by https://github.com/UltimateHackingKeyboard/agent/issues/2080 . (I give 3:1 odds that the code is correct.)
closed 2023-09-06 by mondalaci