UltimateHackingKeyboard/firmware #680: [Feature proposal] Scoping, and control flow

kareltucek, 2023-09-05:
As a followup to named variables, I would like to propose following new features, since working with control flow that consists of more than one command per basic block is pain:

- Introduce `else COMMAND` and `while EXPRESSION COMMAND` conditions. (This is a straightforward feature.)
- Introduce scoping, using curly braces to group multiple commands into one. (This will be a bit more laborious, since every scope will need its own state data, effectively creating a stack out of a chain of newly created scope structures.) I.e.: 
```
COMMAND = {
    COMMAND
    COMMAND
    ...
}
```

E.g.:
```
if ($varA < 3) {
    tapKey a
    untoggleLayer    
} 
else {
    tapKey b
}
```
```
if ($varA < 3) ifNotInterrupted tapKey a
else ifShift tapKey b
```
```
while ($varA < 3) {
    tapKey a
    setVar varA ($varA+1)
}
```

mondalaci, 2023-09-05:
Your proposal looks very impressive. I'm not a heavy smart macro user, but I'm sure many forum folks would appreciate these features, and you're welcome to ask around in the forum.

How about RAM usage regarding per-scope states and, in general, regarding these features?

kareltucek, 2023-09-05:
- else / while are cheap. 1-bit per state each
-  scopes depend on how big pool we will want to allocate... ...which depends on how deeply nested macros we expect people to use. Current pool state size should suffice in theory, which would mean approximately 4 additional bytes per state just to keep links between those structures (64 bytes total, given current 16 macro slots),  since per-scope data will be refactored out of current macro states. But to be safe, I would probably prefer to provide two scope-slots per each macro slot. Assuming 16 bytes per scope, that would mean additional 256+64 bytes...

mondalaci, 2023-09-06:
Thanks! The memory consumption seems reasonable. Feel free to go ahead with this.

PR #704 (2023-10-01): Implement scopes.
Closes #680.

(Please, construct your own test cases, and look into docs if needed. If things are not intuitive or easy to figure out from docs, then something is wrong and needs to be fixed...)
closed 2023-10-01 by mondalaci