UltimateHackingKeyboard/firmware #682: Macro language: special characters.

kareltucek, 2023-09-10:
We also still support both `#` and `//` comment styles, and as far as I remember I promised to discuss and rectify this ambiguity in tome, so here we go.

At the moment, the grammar (documentation) does not explicitly allow comments at the end of line, but the parser accepts them. I believe that allowing such comments is strongly desirable. 

Yet, `#` comments at the end of line are not possible after `tapKeySeq` commands, because `#` is a legal argument to `tapKeySeq`. E.g., `tapKeySeq a b c # d e f # oops, is this a comment, or a continuation of argument list?`

As far as I know this is the only serious problem concerning special characters. Nevertheless, special characters still prevent separating tokeinzation from grammar parsing, which has the following consequences:

- In order to read a token properly, grammar context has to be known.
- Implementing proper syntax highlight in Agent's monaco editor is most likely impossible.
- Using conventional parsing tools to parse the grammar is either impossible or complicated.
- It complicates any idea of compilation of the language into bytecode. (It does not prevent them, but compilation would have to be done using full parsing machinery, instead of let's say a simple pass  that would not care about grammar context.)

I.e., we need to decide what to do about the `#` character. If we at the same time figure out how to eliminate all special characters it will make certain optimization easier, but it is not strictly needed.

@mondalaci what is your opinion on what to do with it? (It is you who insisted on `#` comments over `//`, so I do expect you to have some opinion on this.)

mondalaci, 2023-09-10:
Even though I like `#` comments, given the above, it's probably a good idea to remove them and only allow `//` comments everywhere, including the end of the line.

kareltucek, 2023-09-10:
That is a shame, I too grew fond of them over time.

mondalaci, 2023-09-10:
Me too, but let me know if there's a better way to solve the issue.

kareltucek, 2023-09-10:
Following come to mind:

- Force escaping special characters in commands. E.g., `tapKeySeq \#`, or `tapKeySeq '#'` or something like that.

- Prohibit special characters altogether, and require them to be abbreviated by alphanumeric identifiers.  E.g., `holdKey semicolonAndColon`

- Assert that we don't care and comments after tapKeySeq are not possible.

- Leave both notations in place.

- Make `#` followed by a non-white character act as comments, while those followed by a white character act as an argument.

I am not huge fan of any of the above though. Switching back to '//' indeed makes sense.

mondalaci, 2023-09-10:
I agree that we should only allow using `//` for comments going forward. It's the cleanest solution.

kareltucek, 2023-09-10:
Ok!

kareltucek, 2023-09-11:
Am I correct that I should prohibit the `#` and add deprecation warnings for it?

mondalaci, 2023-09-11:
Yes, please do.

PR #687 (2023-09-11): Unify comment style to // and insist that all macro code is consumed.
Closes #682
closed 2023-09-11 by mondalaci