UltimateHackingKeyboard/firmware #693: Layer indicators glowing even when layer is not active.

kareltucek, 2023-09-15:
Sometimes, my mouse indicator glows faintly, for no reason.

Just now, I have encountered user config from https://github.com/UltimateHackingKeyboard/agent/issues/2083 with which the mouse indicator glows so bright it looks like the layer is always active.

[image]

(I also noticed this mentioned on Discord, but did not pay attention to it at the time.)

Clearly depends on user config.

Seems to depend on keyboard backlight. The indicator is dim with red or green backlight (constant RGB across entire keyboard), but lights up quite brightly with blue backlight.

mondalaci, 2023-09-15:
Are you sure the mouse LED doesn't bleed to fn?

kareltucek, 2023-09-15:
Yes. With green backlight, mouse is dim and fn glows.

(Still not sure if I am clear. In the photo above, mouse layer is NOT active.)

kareltucek, 2023-09-15:
Does it mean that your UHK doesn't do this with functional backlight?

mondalaci, 2023-09-15:
Mine doesn't. I used the mentioned config and couldn't reproduce the issue, but my UHK features an LED display with white LEDs, which can make a difference. Also, older displays might be more prone to this issue. I suspect it's a hardware issue.

mondalaci, 2023-11-04:
Should we close this?

kareltucek, 2023-11-04:
I am not sure.

I am wondering why we have not noticed the problem earlier. I am quite sure we would have noticed if this was the problem from the beginning.

I did try to bisect it to see if it would go away with some earlier firmware version, but it seems to be present independently of firmware. 

(This is problem with v2. My v1 does not suffer by this at all.)

The leds are at times quite disturbingly bright. I consider this to be a problem, and it would be nice to find a way to suppress it. But I have no idea what to do about it, or even how to determine whether it is actually a hardware or software problem.

mondalaci, 2023-11-04:
No one has reported this problem to us yet, which makes me believe it's a hardware problem. We can send you a new display, or if you can't desolder the old one and solder the new one, we'd recall your UHK.

kareltucek, 2023-11-04:
I am not concerned about my UHK. It was discussed on Discord some months ago, so I am quite sure this is not just my problem, and so I treated it as a global problem, which it indeed may not be.

Let's assume that I have indeed exaggerated the scope and close it for now.

closed 2023-11-04 by kareltucek