UltimateHackingKeyboard/firmware #694: 10.2 WRN messages mixed up

rpnfan, 2023-09-17:
The messages I get when printing the error status seems not consistent and also with formatting errors. See the example below.

```

Warning at base - mod layer on 1/1: # comments are deprecated, pl(ASE SWITCH CLICK), hold works as well and does not toggle then.
>   | 
Warning at bed - mouease switch to //
> 1 | # toggle to mouse layer (with one click), hold  worksn as wel base -oespace Mod 1/1: # comments are deprecated, please switch to //
> 1 | # tap: space, ba: m space Xod 1/2: Command deprecated. Please, replace resolveSecondary bY ~IF RY ANCEDStrategy goTo ....
> 2 | resolveSecondary 250 primaryaction se ondarysed encountered.
> 2 | resolveSecondary 250 primaryaction secondaryaction 
>   |          om       e deprecated, please switch to //
> 1 | # A very simple macro that I find and th keys 7,8,9,0
>   | 
Warning at Macro Rec/Play 1/2: # comments are deprecated, please swTCH TO + 7> 2   |# 
Warnin at Macro Rec/Play 1/3: # comments are deprecated, please switch to /
> 3 fn + hift +7 again
>   | 
Warning at Macro Rec/Play 1/4: # comments are deprecated, pleas ketchkes with fn + 7
>   | 
^Warning at Macro Rec/Play 1/6: Registers are now deprED> SETE< R FOO ! AND $foo`.
> 6 | ifShift recordMacro #key
>   |                     
Warning at Macro Rec/e,Play 1/6e #keyegth $thisKeyId.
> 6 | ifShift recordMacro #key
>   |                      
^Warning at Macro Rec/e, play 1/7e thRegh named variables. E.g., `setVar foo 1 and $foo.
`> 7 | ifNE.g., `setVar foo 1 and $foo.
> 7 | ifNE.g., `setVar foo 1 and $foo`.
> 7 | ifNE.g., `setVar foo 1 and `$foo`.
> 7 | ifNE.g., sètVar foo 1 and `$foo`.
> 7 | ifNE.g., setVar foo 1`and $foo`.
> 7 | ifNE.g., `setVar foo 1 and $foo`.
> 7 | ifNE.g., setVar foo 1 and $foo`.
> 7 | ifNE.g., `setVar foo 1`and $foo`.
> 7 | ifNE.g`.`, se`tVar foo 1 and $foo`.
> 7 | ifNE.g., `setVar foo 1 and $foo.
`> 7 | ifNE.g., `setVar foo 1 and `$foo`.
> 7 | ifNE.g., `setVar foo 1 and $foo`.
> 7 | ifNE.g., sètVar foo 1 and $foo`.
`> 7 | ifNE.g., `setVar foo 1 and `$foo`.
> 7 | ifN
```

kareltucek, 2023-09-17:
Please try to use another text editor and let me know the results. My guess is that your editor is confused by super fast input. (It may also trigger autocomplete.)

You can also trigger validation with a `validateUserConfig` command.

rpnfan, 2023-09-22:
I have tried a few editors. Notepad, Notepad2, Word all have the same problem that the text is not complete. When you redo that you will get a somewhat different but always scrambled output. I do not know if there is a way to write to the clipboard on Windows. That might be a workaround?

kareltucek, 2023-09-22:
That's strange. For me output works just fine, unless I forget to set paste mode in vim (disables features like autocomolete or autoindent)... 

As for workarounds, I have no idea if there is a standard way of writing clipboard via usb... ...but it is not relevant as the next agent version will retrieve errors over usb and show them via gui...

kareltucek, 2023-09-22:
As for workarounds you can also slow down the outpu by setting keystrokeDelay to some value like 10 or 50.

rpnfan, 2023-09-23:
Thanks, looking forward the new Agent. A delay of 10 works fine as a workaround. :-)

closed 2023-09-23 by rpnfan