UltimateHackingKeyboard/firmware #700: cannot use setLedTxt to display three dashes (since firmware 10.2.0)

mhantsch, 2023-09-21:
Agent 3.0.2, Firmware 10.2.0.

Macro 'test-test' bound to `Fn`-`t`:
Command: `setLedTxt 200 ---`

When I press `Fn`-`t` to run the macro I get ERR.
Printstatus yields:

```
Error at test-test 1/1: Could not parse
> 1 | setLedTxt 200 ---
>   |                  ^
Error at test-test 1/1: Unexpected variable type: 3
> 1 | setLedTxt 200 ---
```
This used to work in previous firmware. I want to set the LED display to show three dashes.

kareltucek, 2023-09-21:
Please, change the code to `setLedTxt 200 "---"`. 

Sincde 10.2.0., `setLedTxt` accepts also numeric expressions (`setLedTxt $keystrokeDelay` or `setLedTxt (1+2)`, and in this case it interprets three dashes as three unary minuses in a row...

mhantsch, 2023-09-21:
What seems to work is use `setLedTxt 200 "---"` but I don't see in the documentation why I would have to quote the string, nor which quote characters exist and should bu used.

kareltucek, 2023-09-21:
Are you looking into correct docs version? The grammar states:

```
STRING = "<interpolated string>" | '<literal string>'
```

reference manual furthermore states:

> write <custom text> will type rest of the string. Same as the plain text command. Strings are single quote (for literal strings) or double quote (for interpolated strings) enclosed. E.g., `write "keystrokeDelay is $keystrokeDelay, 1+1=$(1+1)\n"`, or `write '$ will show as literal dollar sign.'`.

user guide then states

> Write command accepts (not exactly) bash-style interpolated strings. E.g.:
>
>   ```
>   write "Current keystrokeDelay is $keystrokeDelay. One plus one is $(1 + 1).\n"
>   write 'This is literal "$" character. And here you have an apostrophe '"'"'.'
>   ```

If there is anything I can do to make the docs easier to navigate or search, please let me know!

mhantsch, 2023-09-21:
Sorry, you are right. I was looking at the wrong version of the docs. My browser tab still had an older tag open. My bad.

mhantsch, 2023-09-21:
Resolved by user insight and realisation.

closed 2023-09-21 by mhantsch