UltimateHackingKeyboard/firmware #701: Timeout for OneShot

rpnfan, 2023-09-25:
Add configurable timeout option for OneShot, to prevent "hanging" Shift function when this is used in combination with external mouse combos.

see: https://forum.ultimatehackingkeyboard.com/t/kill-macro-oneshot-after-a-specific-time/494

kareltucek, 2023-09-25:
Thanks for creating the ticket!

Please suggest the default value for the timeout.

rpnfan, 2023-09-25:
>  I looked up what the default repeat delay for a keystroke in Windows is. That is 500ms, with options for 250ms, 750ms and 1000ms as well. 

So for that use case here I think 500 ms is also a good default. For "slow" typing ( about 24 wpm) long enough, but still not too long to have to wait forever, when one is doing another action. For myself I think I want to set it to 250 or 300 ms longest, as holding Shift to mark multiple items in a list will still work and then timing in not critical.

closed 2023-09-26 by mondalaci